bemasher/rtlamr #28: Report radio signal levels

jgarland79, 2015-03-24:
Would it be possible to report all applicable signal level data along with the data for each packet?

bemasher, 2015-03-24:
What sort of data are you interested in getting? It would likely have to be added as a verbose flag since the output is intended to be limited to just decoded message data.

jgarland79, 2015-04-01:
rssi, snr, or whatever you have available to indicate the strength of the signal received

bemasher, 2015-04-01:
SNR would be doable but would require extra computation and can't be guaranteed to be any kind of accurate since this is a fairly wide-band receiver and messages are commonly overlapping other signals. I'll look into adding it.

bemasher, 2015-07-08:
What's the main purpose you'd be using this feature for?

jgarland79, 2015-07-08:
For antenna placement and tuning

bemasher, 2015-09-15:
I'm afraid this feature request will have to be denied. It's not a feature that I believe will be used enough to justify the time to implement it. The idea being that you'll probably only want to do this once for a stationary setup.

My suggestion is to make sample captures using the -samplefile flag and examine them in software such as baudline or audacity to compare received signal strength over variations on antenna placement and orientation. Be sure to disable both the demodulator's AGC and the tuner's AGC before capturing sample packets.

bemasher, 2016-06-29:
I've recently implemented RSSI reporting per another user's request, feel free to give it a try and let me know if it does what you're looking for: 2599ae144b8291600a850bb4b9f1aa788214652f

However, there are a few caveats:
- RSSI varies with frequency relative to the receivable bandwidth. Messages towards the edge of the sampled bandwidth will have smaller RSSI than those near the center. This can be mitigated by setting the center frequency to a known frequency your meter transmits on and using a narrow symbol length: `-symbollength=8` Messages will be received much more infrequently since `rtlamr` is a promiscuous receiver that attempts to receive all messages within it's sampled bandwidth.
- Tuner AGC is enabled by default and will affect the ability to consistently determine RSSI per message. It is recommended to disable this and set the gain manually: `-tunergainmode=false -gainbyindex=29` where `-gainbyindex` is >=0 and <= GainCount reported at startup. This will severely limit sensitivity and the distance from the meter you can reliably receive messages.

closed 2016-10-02 by bemasher