bemasher/rtlamr #29: Forgive my non-issue :(

bcsteeve, 2015-04-29:
Sorry for reporting this as an "issue"... I just can't figure out where I can ask a question.

I have a Neptune R900i water meter in my home.  I was actually excited to learn I would be getting a smart meter because I thought I'd be able to better monitor my water usage, but it seems the extent the features are used is to simply make their meter reading tasks easier.

I want to know if I have a leak, in as close to real-time as possible.

So I started looking for ways to read the meter myself (other than visually looking at it) and that brought me here, but I'm still lacking a LOT of info.

So a) can I interface (RTL-SDR?) with this meter so I can get the reading?  and b) can I get a high enough resolution and frequency to actually make use of the data?

Again, sorry for posting this as an issue.

bemasher, 2015-04-30:
R900 meters are supported and flag information was added in e1d9b04354bdec16a38a2740053c33712f79715d. 

I don't physically have one of these meters so I'm not especially familiar with the meaning of each flag. I'll see if the contributor of that commit can clarify these for you and I'll get it added to the documentation.

bemasher, 2015-04-30:
This is taken from e1d9b04354bdec16a38a2740053c33712f79715d and formatted for inclusion in documentation later.

## Research
From public documentation: https://www.neptunetg.com/About/Case-Studies/Water/Westbank-Irrigation-District,-BC

It indicates that the utility can gather stats on days of no use and days of leak in the past 35 days, and that it is divided into buckets as shown later.

Using this public information, intentionally triggering the leak flag in my home, and listening to many meters over time led me to this decoding of the `Unkn2` and `Unkn4` fields.
 
`?`'s are still unknown, but this patch lumps them in with the bucket number.

### `Unkn2`: NoUse and Backflow

The 8 bits of `Unkn2` seem to show days of no use and backflow information, like so:
 
	???BBBHL

Symbol | Meaning
------:|:-------
   BBB | Bucket # for days of no use in past 35 days.
   H   | High backflow in past 35 days.
   L   | Low backflow in past 35 days.

Backflow seems to require some minimum frequency of occurrence before it will set the flag.

### `Unkn4`: Leak and LeakNow

Similarly, the 6 bits of `Unkn4` look like this:

	?BBBHL

Symbol | Meaning
------:|:-------
   BBB | Bucket # for days of leak in past 35 days.
   H   | Continuous leak in past 24 hours.
   L   | Intermittent leak in past 24 hours.

Intermittent leaks are described by Neptune in other public docs as: "Intermittent leak indicates that water has been used for at least 50 of the 96 15-minute intervals during a 24-hour period". A continuous leak indicates use in every 15-minute period over the past 24 hours.

## Implementation

Plaintext output now shows:

	NoUse:$BIN BackFlow:$FLAG ... Leak:$BIN LeakNow:$FLAG

`BackFlow` covers the last 35 days. `LeakNow` covers the last 24 hours. Each has one of three values:

Value | Meaning
-----:|:-------
	0 | None
	1 | Low
	2 | High

`NoUse` and `Leak` are numbers between 0 and 6 that represent the number of days within the last 35 days the condition has occured:

Value | Days
-----:|:----
    0 | 0
    1 | 1  -  2
    2 | 3  -  7
    3 | 8  - 14
    4 | 15 - 21
    5 | 22 - 34
    6 | 35+

closed 2015-06-06 by bemasher