bemasher/rtlamr #31: rtltcp does not build, no error

dennisholt, 2015-05-08:
I try to build rtltcp go seems to complete normally but rtltcp is not created in $GOPATH/bin
I am running Apple OS 10.10.3 with 'go' version 1.4.2 
I was able to build rtlamr and a "hello" program added to github.com/bemasher/ 
The following lines are from the terminal attempt to build rtltcp. I also echoed $GOPATH and $PATH fyi:
     MacBook:src jeanie$ go install github.com/bemasher/rtltcp
     MacBook:src jeanie$ rtltcp
     -bash: rtltcp: command not found
     MacBook:src jeanie$ echo $GOPATH
     /Users/jeanie/dennis/Computer/go
     MacBook:src jeanie$ echo $PATH
     /opt/local/bin:/opt/local/sbin:/opt/local/bin:/opt/local/sbin:/opt/local/bin:
        /opt/local/sbin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/bin:/opt/X11/bin:
        /usr/local/git/bin:/usr/local/go/bin:/Users/jeanie/dennis/Computer/go/bin
I can run rtlamr but without rtltcp I get the following:
     MacBook:src jeanie$ rtlamr
     17:10:51.020337 decode.go:42: BlockSize: 4096
     ......    8 lines removed for brevity  ......
     17:10:51.020512 decode.go:51: Preamble: 111110010101001100000
     17:10:51.020781 recv.go:62: Error connecting to spectrum server: dial tcp 127.0.0.1:1234: connection          refused

However, if I run osmo.com rtl_tcp then run rtlamr things appear to work with the following:
MacBook:src jeanie$ rtlamr
     17:34:46.350806 decode.go:42: BlockSize: 4096
      ......    8 lines removed for brevity  ......
     17:34:46.351097 decode.go:51: Preamble: 111110010101001100000
     17:34:46.351560 recv.go:69: GainCount: 29
     {Time:2015-05-08T17:34:46.642 SCM:{ID:34211079 Type: 4 Tamper:{Phy:00 Enc:01} Consumption:   38992 CRC:0x6E79}}
     {Time:2015-05-08T17:34:47.085 SCM:{ID:34237682 Type: 4 Tamper:{Phy:00 Enc:02} Consumption:   89619 CRC:0x272B}}
     {Time:2015-05-08T17:34:47.739 SCM:{ID:34211079 Type: 4 Tamper:{Phy:00 Enc:01} Consumption:   38992 CRC:0x6E79}}
     {Time:2015-05-08T17:34:47.959 SCM:{ID:34237682 Type: 4 Tamper:{Phy:00 Enc:02} Consumption:   89619 CRC:0x272B}}  

Any hints on how to get rtltcp to build and run?  I am thinking this will work more like designed if rtlamr has rtltcp to talk to and set-up rtl_tcp.  Thanks for a writing this program and introducing me to 'go'.  Looks like an interesting language.  
Best regards,  Dennis

bemasher, 2015-05-08:
You're confusing the package `github.com/bemasher/rtltcp` that allows `rtlamr` to communicate with the TCP spectrum server from the osmocom tools named `rtl_tcp`. The package when compiled won't produce an executable, it is only used internally by `rtlamr`.

I've chosen not to encapsulate execution of `rtl_tcp` in Go because I'm not prepared to test or support doing so on multiple platforms. The current solution allows the best flexibility and control without reproducing functionality.

closed 2015-05-08 by bemasher

dennisholt, 2015-05-08:
Douglas,
Thanks for the very quick answer.  This has been a slow but good learning experience so far.  Thanks for writing and supporting rtlamr.  It makes learning this stuff much easier for us neophytes. 
Dennis

bmwilson55, 2017-08-02:
Hello. I am receiving the same "error connecting to spectrum server" that Dennis was getting.  How is this resolved? Thanks

dennisholt, 2017-08-02:
It has been a while since I got this set-up but it is working well at this time.  Here is what I do: 
1. Plug the sdr dongle into the usb
2. I open a terminal window and run: rtl_tcp.  This sets up the radio and provides a service where rtlamr can get data.  I wait until device is found and the set-up characteristics are printed to the terminal.
3. I open another terminal window and run: /Users/jeanie/Dennis/Computer/go/bin/rtlamrh -format=plain -msgtype=scm -filterid=3062204,3062205,3062206   (I modified rtlamr and named it rtlamrh)
It can take a minute or so before the radio picks up transmission from my meter.  
4. After I get the data I want I first stop rtlamr using Ctrl-C then close the terminal window.  Next stopping rtl_tcp and closing its terminal session.

Hope this helps.  If you still have problems: make sure you have downloaded the latest version of go.  I found that fixed a problem that cropped up when my OS was updated recently.  
Dennis

bmwilson55, 2017-08-02:
Excellent! I think it is working now, however I can't see any data coming in.  I assume it appears in the terminal session.  I removed the filterid but still can't see data.  Here is the output of my rtlamr.  Any thoughts? Thanks for your help in getting it started!

21:26:32.303957 decode.go:79: CenterFreq: 912600155
21:26:32.304238 decode.go:79: SampleRate: 2359296
21:26:32.304292 decode.go:79: DataRate: 32768
21:26:32.304341 decode.go:79: ChipLength: 72
21:26:32.304391 decode.go:79: PreambleSymbols: 21
21:26:32.304440 decode.go:79: PreambleLength: 3024
21:26:32.304488 decode.go:79: PacketSymbols: 96
21:26:32.304536 decode.go:79: PacketLength: 13824
21:26:32.304583 decode.go:79: Preamble: 111110010101001100000
21:26:32.304630 main.go:93: GainCount: 29

dennisholt, 2017-08-02:
The output you posted indicates you are connecting to the rtl_tcp.  Otherwise the Gain count doesn't show up and there is an error message.  However some of the settings are different from mine.  I don't know if my version of rtlamr is the most recent but it works for reading my meter.  Here are the settings posted by my version:
19:43:01.144921 decode.go:42: BlockSize: 4096
19:43:01.145116 decode.go:43: CenterFreq: 912600155
19:43:01.145126 decode.go:44: SampleRate: 2392064
19:43:01.145133 decode.go:45: DataRate: 32768
19:43:01.145139 decode.go:46: SymbolLength: 73
19:43:01.145144 decode.go:47: PreambleSymbols: 21
19:43:01.145150 decode.go:48: PreambleLength: 3066
19:43:01.145155 decode.go:49: PacketSymbols: 96
19:43:01.145160 decode.go:50: PacketLength: 14016
19:43:01.145165 decode.go:51: Preamble: 111110010101001100000
19:43:01.145562 recv.go:71: GainCount: 29
So I am guessing that your may need to change some settings.  Also try taking out the -msgtype=scm. When I first got this working it reported a lot of meters in the area.  Not all that many on scm.
Good Luck,
Dennis

bmwilson55, 2017-08-03:
Hmm. So I adjusted the symbollength and my numbers look like yours now.  What rtl-sdr radio are you using?  I tried taking out the msgtype with no luck.   The odd thing is, I have been able to use my sdr as an FM radio so I know its connected up right.  rtl_tcp also appears to give notice that a connection is made, just no data.  Did you have to set anything specific for it to receive your specific meter type? Wondering if my meter is for some reason not sending signals....

02:32:13.078228 decode.go:79: CenterFreq: 912600155
02:32:13.078519 decode.go:79: SampleRate: 2392064
02:32:13.078573 decode.go:79: DataRate: 32768
02:32:13.078645 decode.go:79: ChipLength: 73
02:32:13.078697 decode.go:79: PreambleSymbols: 21
02:32:13.078746 decode.go:79: PreambleLength: 3066
02:32:13.078794 decode.go:79: PacketSymbols: 96
02:32:13.078869 decode.go:79: PacketLength: 14016
02:32:13.078919 decode.go:79: Preamble: 111110010101001100000
02:32:13.078968 main.go:93: GainCount: 29

rtl_tcp output:

client accepted!
set freq 912600155
set sample rate 2392064
Exact sample rate is: 2392064.101227 Hz
set gain mode 0

bemasher, 2017-08-03:
You shouldn't need to adjust the symbollength in order to receive messages that will only affect performance and receivable bandwidth.

Removing -msgtype=scm doesn't do anything because if -msgtype isn't specified, it defaults to scm. You'll want to try the other msgtypes available, they are: scm, scm+, idm, r900 and r900bcd. Your meter may not be ERT compatible.

bmwilson55, 2017-08-03:
Thanks bemasher and dennis for the info.  So, after several hours of attempts, I believe my meter is the problem.  I have an itron centron C2M that has OpenWay.  From what I read elsewhere it looks like this meter uses ZigBee Smart Energy as its  comms.  What I can't understand is if it still broadcasts SCM/IDM at 915mhz like the other meters do.  I am thinking it does not since I am not seeing data on my rtlamr.  Any experience with these meters?  Pretty bummed at this point.

dennisholt, 2017-08-03:
My meter is a GE 'G0203 Net Meter' it has a Itron communication board FCC ID: CDT5XESS.  
Your meter should have a FCC ID on it somewhere and you can look that up in the FCC database which will give you the frequency and other info.  My recollection is that OpenWave is encrypted.  There are probably others working on it and you can search around on the internet to see what you can find.