bemasher/rtlamr #33: Let us compile without GoLang

michelinok, 2015-06-22:
Hi!
Is there a way to compile without GoLang? (maybe with Make ?).
It would be great for those like me that uses Raspberry and wanna try your software.
(GoLang is very very heavy!)

danielschudel, 2015-06-22:
I'm using this repo for Go for RPi, not sure if you know about it:

pi@sdrpi20 /etc/apt/sources.list.d $ cat $PWD/xivilization-raspbian.list 
deb http://xivilization.net/~marek/raspbian/ wheezy main non-free
deb-src http://xivilization.net/~marek/raspbian/ wheezy main non-free

You are right about Go being heavy. I do experience missing utility reading broadcasts. But can usually count on getting 3-4 updates/hour, which is frequent enough for me.

And I am using a RPi 2 Model B.

bemasher, 2015-06-22:
The project was written in Go, so no, you can't compile it without the Go toolchain. The easiest way to get the go tool-chain is to download one of the unofficial releases for your particular model from: 

http://dave.cheney.net/unofficial-arm-tarballs

Another easier option is to simply download one of the binary releases:

https://github.com/bemasher/rtlamr/releases

 * RPi: ARMv6
 * RPi 2: ARMv7

As far as Go being "heavy" I'm not sure I know what you mean. Go produces pretty good code and I've optimized the most expensive operations pretty heavily.

You won't be able to run rtlamr at full bandwidth on the original RPi without dropping samples, it simply isn't powerful enough for that. The least CPU intensive (and narrowest bandwidth) configuration is as follows:

    rtlamr -symbollength=8 -decimation=2 [other flags]

I've tested on the original RPi with the following settings and so long as it is only running rtl_tcp and rtlamr, it drops very few samples. The RPi 2 should have no trouble at all running this:

    rtlamr -symbollength=72 -decimation=18 [other flags]

closed 2015-06-22 by bemasher