bemasher/rtlamr #36: R900 meter readings

chaplijt, 2015-08-05:
have been reading 6 R900 water meters with your software using a Raspberry Pi  and have collected a bunch of data. Unfortunately, it appears that the electronic meter readings do not correspond with actual meter consumption. (realize there may be an offset between physical meter and electronic, but consumption amount (change between two dates)  should match up.

Seem to remember you didn't have access to a meter. Any idea how to resolve this discrepancy?  I can get more data as needed.

john

data results:

[image]

. Actual physical meter readings					
11-Jul	11:25	82240.5	47624.2	87430.1	84552.7
5/2/2015		        80997.7	45955.1	85285.3	84083.3
					
delta	cf	                  1242.8	1669.1	2144.8	469.4
	ccf	                  12.428	16.691	21.448	4.694
					
electronic readings (approximately same time as data above)					
11-Jul		        533040	292384	554032	542032
5/2/2015		        526741	285013	545552	540800
					
delta	cf	                   6299	7371	8480	1232
	ccf                      62.99	73.71	84.8	       12.32
					
ratio		                        5.1	4.4	            4	          2.6

bemasher, 2015-08-07:
If you could provide a verbatim log from one of your collection runs that would be most helpful. You can do so privately if you like.

chaplijt, 2015-08-07:
DO you want raw or the data collection file?
john

chaplijt, 2015-08-07:
data sent from steveswatermeters

bemasher, 2016-03-08:
It appears that certain r900 meters store the consumption value as a binary-coded-digit. I've been looking at creating a decoder to handle this. I'll let you know when it is complete.

bemasher, 2016-04-10:
I've implemented the r900bcd protocol, if you could test it out for me that would be fantastic since I don't have access to an r900 capable meters.

If you don't have the go build environment setup, let me know what platform you're running this on and I can build a binary for you to test.

leomrlima, 2016-04-11:
Hello! I can test it in Linux x86 and Mac OS x64. Although I've been getting correct readings from the meter I have access...

bemasher, 2016-04-12:
This tarball contains both darwin/amd64 and linux/386 executables for the r900bcd branch. If you could post a log of several messages, that'd be great. Currently there is no sanity check to be sure that your meter isn't transmitting non-bcd messages and will likely output 0's if the hexadecimal consumption contains any of the characters A-F.

[r900bcd.tar.gz](https://github.com/bemasher/rtlamr/files/214312/r900bcd.tar.gz)

leomrlima, 2016-04-22:
I tested with my meter and the values are all 0

It seems I don't have a BDC enabled meter?

{Time:2016-04-22T20:46:30.919 Offset:0 Length:49792 R900:{ID:         0 Unkn1:0x00 NoUse: 0 BackFlow:0 Consumption:       0 Unkn3:0x00 Leak: 0 LeakNow:0}}

More data in the attached file.
[Archive.zip](https://github.com/bemasher/rtlamr/files/232548/Archive.zip)

bemasher, 2016-04-22:
As it turns out the point in the master branch this branch was based on had an R900 bug, which has since been fixed. I've merged the fix into this branch and recompiled the two binaries, if you could try the same procedure again that'd be great.

[r900bcd.tar.gz](https://github.com/bemasher/rtlamr/files/232675/r900bcd.tar.gz)

chaplijt, 2016-05-19:
Was able to check out the r900bcd option and it seems to work for the meters I tested (comparison data in attached excel file.)

Seems like the unkn1 field holds a meter version number:
A1 is the mechanical meter pictured in the excel sheet. it uses the BCD encoding r900 BCD option
A3 is an LCD display meter that sends an  integer value which is read properly by the r900 option.

john
[r900 compare.xlsx](https://github.com/bemasher/rtlamr/files/273276/r900.compare.xlsx)

chaplijt, 2016-05-19:
btw, the electronic meter reading of 33225 versus the picture of the meter at 33227 is due to time lapse. Took some time to find a meter I could get clear picture of. also meter is for gym, so that amount of consumption seems reasonable.
john

bemasher, 2016-05-19:
Looks good, I'll make sure this goes into the next release. It may be possible for R900 meters only to switch this automatically based on the field values like that.

I'm seriously considering splitting R900 decoding into it's own project since it complicates the structure and organization for all of the signal processing that will need to change to support multiple protocols per invocation.

closed 2016-05-19 by bemasher

cypherstream, 2016-07-08:
How do you determine your meter vs others in the neighborhood?  I went to my water bill online account and my meter number does not match any of the ID's that is being reported.  I get more values using r900 but there seem to be some out in the neighborhood using r900bcd which I think its the mechanical meter.  Mine is a Neptune meter with an LCD on it that activates when you shine a flashlight on the light sensor on the meter.

kb1ca, 2016-07-08:
Transmissions from Neptune E-Coder R900 units are identified by a MIU (Meter Indicating Unit) serial number. It's a 10 digit number that begins with a 1 and the first three digits indicate the model series. The different model series have different data logging and fault indicating bits (backflow, leak, historical data etc.), and also appear to use different methods of encoding the register value (eg. a 185 meter uses the 'standard' method).

Be aware that the value transmitted by the indicating unit typically have one less digit than the value being displayed on the meter display - the least significant digit is typically dropped as it's unnecessarily precise for billing purposes. The degree of precision is used internally for leak and backflow detection, hence appearing on the meter display.

If your meter is an 'integrated' model where the meter register and indicating unit transmitter are contained in one housing then the MIU is either laser etched on the outside of body, or stamped on the meter face behind the glass. The meters ship with an FCC certification label affixed with the same information shown on a code39 barcode however the sticker can easily be damaged or removed.

This is an example of a 154 series integrated water meter where the MIU serial number is etched on the case: http://d2ydh70d4b5xgv.cloudfront.net/images/2/a/one-1-new-neptune-digital-water-meter-e-coder-r900i-rw2f23-r900m-t10-3-4-t-10-6b0eea19da979323cb3b7ed31baa9295.jpg

This is a 183 series integrated water meter where the MIU serial number is stamped: https://i.ytimg.com/vi/INDEHZzX-Rw/maxresdefault.jpg

This is a 148 series meter indicating unit that connects to and transmits values from an externally connected meter register: http://jimlaurwilliams.org/wordpress/wp-content/uploads/2012/09/NeptuneR900-3543.jpg   This particular example indicating unit also transmits two separate MIUs (1484093176 and 1484093177), the combination of the two MIUs is used to convey larger register values for high flow services (an example configuration might be 10's of cubic meters via one MIU and litres via the other). The two separate readings are combined in the billing software during post-processing to derive a single consumption figure which would be shown on the invoice.