bemasher/rtlamr #37: Any chance of supporting ELSTER R2S meters?

xphyr, 2015-10-13:
So I hate to ask this, but is there any chance you might add support for the ELSTER R2S meters?  I have found code that can be used with GNU radio to read the meters, but the code is well beyond my ability to port it.  The code is on github under argilo/gr-elster.  Thanks for sharing this code and love the fact that you used Go to do it in.

xphyr, 2015-10-13:
Never mind, I did a lot more research about the specific model I have, it does NOT transmit these R2S signals.  Unfortunately it would appear that they installed a non-remote query-able device.  Thanks for the great work on this application.

closed 2015-10-13 by xphyr

bemasher, 2015-10-13:
Unfortunately Elster R2S meters use FSK modulation which isn't easily added to rtlamr. They do however use TI's CC1110 transceiver which is easy to find. You could get one of TI's CC1111 USB dongles and write some firmware to receive the messages.

Frequency: 902.4 - 927.6 MHz
Channels: 25
Channel Spacing: 400 kHz
Modulation: FSK
Data Rate: 35.5 kbps or 142.2 kbps

http://www.ti.com/tool/cc1111emk868-915

If you aren't interested in writing firmware from scratch (USB is painful), check out: https://bitbucket.org/atlas0fd00m/rfcat