bemasher/rtlamr #39: Support both IDM and SCM messages in one invocation

jepler, 2015-12-08:
I happen to have a supported electric meter and water meter, but it appears that one sends IDM messages and the other SCM messages.  Is it only a software limitation that the two distinct message types can't be received and decoded by a single rtlamr instance?

Right now my workaround is to invoke it in single shot mode first for one message type and then the other, and that does work so woo

bemasher, 2015-12-08:
It's primarily a software limitation at the moment. It is theoretically possible but messy with the way things are structured now. It is also a matter of performance, you almost certainly won't be able to do this on something like a Raspberry Pi.

I'll do some investigation into how easily this could be done.

jepler, 2015-12-08:
Indeed, my solution over in pull request #40 for scm+ism is taking ~100% of one CPU thread on the Odroid U3 where it is running--so close to 100% that now I wonder if it isn't dropping packets to boot.  I didn't think to look at CPU usage when running in just scm or just ism mode, but I can see that splitting the work in the wrong place would double resource usage.

bemasher, 2015-12-09:
It's also a matter of coordinating the two receivers. One solution I've considered is building a multiplexed rtl_tcp clone in go, you get all the same features, and the ability to connect multiple clients who will all receive the same data.

parkerlreed, 2017-09-27:
This is currently working for me using https://github.com/simonyiszk/rtl_mus

[image]

Top right is rtlamr is default scm and bottom right is idm. Both are pointed to the 7373 port created by rtl_mus. I receive messages on both.

bemasher, 2018-05-28:
I've just committed multi-protocol support to a new feature branch. If anyone would be willing to test this, let me know how it goes. You can install the new branch as follows:

```bash
git fetch
git checkout multiprotocol
go install -v .
rtlamr -msgtype=all
```

This feature will require some documentation before it can be included in the next release.

cypherstream, 2018-05-30:
I tried a few options for multiprotocol and none work

```
c:\Go\bin\bin>rtlamr -msgtype=all
17:10:48.611308 main.go:55: invalid message type: "all"

c:\Go\bin\bin>rtlamr -msgtype=scm,scm+,idm
17:11:30.454886 main.go:55: invalid message type: "scm,scm+,idm"

c:\Go\bin\bin>rtlamr -msgtype=scm,idm
17:11:34.517369 main.go:55: invalid message type: "scm,idm"

c:\Go\bin\bin>rtlamr -msgtype= ?
17:12:47.464294 main.go:55: invalid message type: ""
```

This is a fresh install from running C:\go\bin>go get github.com/bemasher/rtlamr

Your command up there doesn't make sense on windows.  
git fetch
fatal: not a git repository (or any of the parent directories): .git

ekrohn, 2018-05-30:
cypherstream,

Odds are you built master, rather than the multiprotocol
branch. If so, when you run rtlamr -help you see:
  -msgtype=scm: message type to receive: scm, scm+, idm, r900 and r900bcd
rather than the newer and longer:
  -msgtype=scm: comma-separated list of message types to receive: all,
scm, scm+ , idm, netidm, r900 and r900bcd                

Douglas,

Is there a reason -msgtype all does not include netidm? At least that's
what this go-ignorant programmer sees in main.go.

--
Eric Krohn

On Wed, 30 May 2018 14:16:22 -0700, cypherstream
<notifications@github.com> wrote:

I tried a few options for multiprotocol and none work

```
c:\Go\bin\bin>rtlamr -msgtype=all
17:10:48.611308 main.go:55: invalid message type: "all"

c:\Go\bin\bin>rtlamr -msgtype=scm,scm+,idm
17:11:30.454886 main.go:55: invalid message type: "scm,scm+,idm"

c:\Go\bin\bin>rtlamr -msgtype=scm,idm
17:11:34.517369 main.go:55: invalid message type: "scm,idm"

c:\Go\bin\bin>rtlamr -msgtype= ?
17:12:47.464294 main.go:55: invalid message type: ""
```

This is a fresh install from running C:\go\bin>go get
github.com/bemasher/rtlamr

Your command up there doesn't make sense on windows.  
git fetch
fatal: not a git repository (or any of the parent directories): .git

cypherstream, 2018-05-31:
Do you have any how to get the multiprotocol one compiled for Windows x64?  I'm really struggling to figure out how to use go.  I *THINK* I got GitHub Desktop to sync the rtlamr-multiprotocol but I have no idea where to put the files.

A step by step would help if you can't just upload the .exe somewhere.

bemasher, 2018-05-31:
@ekrohn netidm is not included in `all` because it is the same preamble/length/checksum as idm, it only interprets the contents of the packet differently. The idea behind `all` is that it should at least show messages from each protocol, and users should only be using it for discovery. Once they know what is being transmitted, they should select the appropriate message types based on their meter(s). This is the same reason `r900bcd` isn't included in `all`.

bemasher, 2018-05-31:
@cypherstream you may want to try using git clone instead of go get.

From a Git Bash terminal:
```
cd %GOPATH%/github.com/bemasher
rm -rf rtlamr
git clone https://github.com/bemasher/rtlamr.git
cd rtlamr
git checkout multiprotocol
go install -v .
```

cypherstream, 2018-05-31:
Seems a bug in Windows 10 when you set the GOPATH env variable, it adds a semicolon at the end.  If you edit the text of this variable and remove the semicolon, you finally stop getting errors on the GOPATH.

Though I'm not sure how to install it.  I've used github desktop and changed to the multiprotocol branch.  It dropped all of this in a folder in my documents.  I copied this into my my documents\go folder (where i set my gopath).  I'm getting this error now after running install -v

C:\Users\cypherstream\Documents\go>go install -v
go install: no install location for directory C:\Users\cypherstream\Documents\go outside GOPATH
        For more details see: go help gopath

cypherstream, 2018-05-31:
Thanks I think we crossed paths there.

Ok so I'm trying your method and the files downloaded but I think the real issue I'm having is with go.

`C:\users\cypherstream\Documents\go\bemasher\rtlamr [multiprotocol ≡]> go install -v
flags.go:31:2: cannot find package "github.com/bemasher/rtlamr/csv" in any of:
        C:\Go\src\github.com\bemasher\rtlamr\csv (from $GOROOT)
        C:\Users\cypherstream\Documents\go\src\github.com\bemasher\rtlamr\csv (from $GOPATH)
main.go:35:2: cannot find package "github.com/bemasher/rtlamr/idm" in any of:
        C:\Go\src\github.com\bemasher\rtlamr\idm (from $GOROOT)
        C:\Users\cypherstream\Documents\go\src\github.com\bemasher\rtlamr\idm (from $GOPATH)
main.go:36:2: cannot find package "github.com/bemasher/rtlamr/netidm" in any of:
        C:\Go\src\github.com\bemasher\rtlamr\netidm (from $GOROOT)
        C:\Users\cypherstream\Documents\go\src\github.com\bemasher\rtlamr\netidm (from $GOPATH)
flags.go:32:2: cannot find package "github.com/bemasher/rtlamr/protocol" in any of:
        C:\Go\src\github.com\bemasher\rtlamr\protocol (from $GOROOT)
        C:\Users\cypherstream\Documents\go\src\github.com\bemasher\rtlamr\protocol (from $GOPATH)
main.go:37:2: cannot find package "github.com/bemasher/rtlamr/r900" in any of:
        C:\Go\src\github.com\bemasher\rtlamr\r900 (from $GOROOT)
        C:\Users\cypherstream\Documents\go\src\github.com\bemasher\rtlamr\r900 (from $GOPATH)
main.go:38:2: cannot find package "github.com/bemasher/rtlamr/r900bcd" in any of:
        C:\Go\src\github.com\bemasher\rtlamr\r900bcd (from $GOROOT)
        C:\Users\cypherstream\Documents\go\src\github.com\bemasher\rtlamr\r900bcd (from $GOPATH)
main.go:39:2: cannot find package "github.com/bemasher/rtlamr/scm" in any of:
        C:\Go\src\github.com\bemasher\rtlamr\scm (from $GOROOT)
        C:\Users\cypherstream\Documents\go\src\github.com\bemasher\rtlamr\scm (from $GOPATH)
main.go:40:2: cannot find package "github.com/bemasher/rtlamr/scmplus" in any of:
        C:\Go\src\github.com\bemasher\rtlamr\scmplus (from $GOROOT)
        C:\Users\cypherstream\Documents\go\src\github.com\bemasher\rtlamr\scmplus (from $GOPATH)
main.go:33:2: cannot find package "github.com/bemasher/rtltcp" in any of:
        C:\Go\src\github.com\bemasher\rtltcp (from $GOROOT)
        C:\Users\cypherstream\Documents\go\src\github.com\bemasher\rtltcp (from $GOPATH)`

cypherstream, 2018-05-31:
Ok I relize its looking for a folder under go called src.  So I copied the files that git downloaded into a new folder I created under Documents\go called src\github.com

Then it found everything but rtltcp, so I took the one that I had from my old working version and pasted it in the same patch Documents\go\bemasher\rtltcp.

Now it no longer complains about any files missing but gives me this.

`go install: no install location for directory C:\Users\cypherstream\Documents\go\bemasher\rtlamr outside GOPATH
        For more details see: go help gopath`

I'm getting closer.  Going to keep banging away at it.  Its been quite some time since I built the original, and I was on an earlier windows version back then.  Last week I updated to Windows 10 1803 so not sure but I didn't have Git or Go at all.  I had to install both of those today.

cypherstream, 2018-05-31:
Ok never mind my last few posts.  Quite a few people had that issue.  Found my way through it from a post over at stackexchange.  It was for unix based system but basically an environment variable called GOBIN needed to be set.  I did this and then it was able to build properly.

I am now able to run it with the -msgtype=all so I can discover what type of protocols I am finding in our area.  We've had remote read water and gas meters and just yesterday they put on an Itron OpenWay IP Mesh digital electric meter.  I wanted the ability to search through all message types to determine which devices used which protocols.

bemasher, 2018-05-31:
@cypherstream I'm going to hide a few of your posts between your last one and this one to keep the issue on topic.

bemasher, 2018-09-02:
Preliminary support has been added in release v0.9.0

closed 2018-09-02 by bemasher