bemasher/rtlamr #44: No -version info

KanyonKris, 2016-01-27:
When I install rtlamr with:
go get github.com/bemasher/rtlamr

and run:
rtlamr -version

all of the fields are undefined (no version info).

bemasher, 2016-01-27:
Unfortunately this is by design. The values for build date and commit hash are set at compile time due to the VCS chicken-and-egg problem. I'll post the build parameters later tonight.

bemasher, 2016-01-28:
From the rtlamr directory the following command will build the command with version info inserted by the linker:
```
go install -v -ldflags "-X main.buildDate=`date -u '+%m-%d-%Y'` -X main.commitHash=`git describe --tags --always`"
```
This should also work for `go build` if you want to build a binary in the current directory instead.

closed 2016-01-28 by bemasher