bemasher/rtlamr #46: Not receiving messages

nklugman, 2016-03-06:
Hi,

I followed the readme and believe I have everything set up correctly. I am trying to listen to an Itron SK9AMI6 meter using a RTL2838UHIDIR. I am running rtlmar without arguments. I have also tried all message types and get CenterFreq, SampleRate, DataRate, SymbolLength, PreaambleSymbols, PreambleLength, PacketSymbols, PacketLength, Preample and GainCount printouts, but no messages. Any tips? 

Thanks!
~Noah

bemasher, 2016-03-06:
The SK9AMI6 has both a 900 MHz radio and a 2.4GHz ZigBee radio. The 900 MHz radio would be the one that needs to be used in order to receive messages. Unfortunately the 900 MHz radio supports both OOK and FSK protocols and rtlamr supports only the OOK variant of ERT.

If you follow the instructions here: https://github.com/bemasher/rtlcap and make a 10MB capture using the squelch option as described in the documentation I can take a look at the capture to see what's likely being used in your area. Use `-centerfreq=915M` and `-samplerate=1382400`

nklugman, 2016-03-06:
Thanks for getting back to me so quickly. Fingers crossed my meter is running OOK.

I followed the instructions above. First I ran: 
`./bin/rtlcap -centerfreq=915M -samplerate=1382400 -tunergainmode=false`
which gave me:
```
12:38:14.818493 Min: 0.000 Max: 3.124
12:38:15.767028 Min: 0.682 Max: 0.696
12:38:16.809729 Min: 0.682 Max: 0.695
12:38:17.757909 Min: 0.680 Max: 0.694
12:38:18.801079 Min: 0.682 Max: 0.696
12:38:19.843726 Min: 0.679 Max: 0.694
12:38:20.792018 Min: 0.682 Max: 0.693
12:38:21.835120 Min: 0.682 Max: 0.696
12:38:22.783051 Min: 0.680 Max: 0.694
12:38:23.825995 Min: 0.683 Max: 0.695
12:38:24.773973 Min: 0.681 Max: 0.694
```

From this I found a squelch of .689

I then ran:
`./bin/rtlcap -centerfreq=915M -samplerate=1382400 -tunergainmode=false -squlch=0.689 -bytes=10M -o cap`
which gave me:
```
12:44:18.521139 Min: 0.000 Max: 0.734
12:44:19.564017 Min: 0.682 Max: 0.695
12:44:20.512011 Min: 0.681 Max: 0.695
12:44:21.554921 Min: 0.682 Max: 0.695
12:44:22.503033 Min: 0.682 Max: 0.694
12:44:23.546101 Min: 0.683 Max: 0.695
12:44:24.494132 Min: 0.681 Max: 0.696
12:44:25.537096 Min: 0.680 Max: 0.694
12:44:26.485233 Min: 0.683 Max: 0.694
```
I am attaching a zip of the cap file below that was generated from this run.

I tried a fresh install of everything this morning and still no luck with rtlamr. 
Thank you again for your help!

[cap.zip](https://github.com/bemasher/rtlamr/files/160476/cap.zip)

bemasher, 2016-03-07:
Try again with squelch level set at 0.7, there was nothing at all in your capture but noise.

nklugman, 2016-03-07:
I'm attaching the new cap with .7 squelch. Thanks. 
[cap2.zip](https://github.com/bemasher/rtlamr/files/161069/cap2.zip)

bemasher, 2016-03-08:
Unfortunately there was still nothing of note in the capture. I just realized you've disabled the tuner's AGC but haven't set any gain. I'd suggest skipping the -tunergainmode flag and determine a new squelch level. Keep in mind that you want the squelch level to ignore the majority of the Max values and keep only the short bursts.

nklugman, 2016-03-12:
I ended up leaving this on for awhile and finally got a packet. Maybe the meters around me are programmed to transmit infrequently. Anyways... I'll close my issue. Thank you for all of your help, pretty cool that you were willing to go through those raw dumps for me.

closed 2016-03-12 by nklugman