bemasher/rtlamr #47: Not receiving R900 readings after 0.7 update

leomrlima, 2016-03-07:
Hello! 

After updating from 0.6.2 to 0.7, I can't receive any more R900 data. I validate that the same flags work on 0.6.2 and not 0.7 running them one after another.

I'm using -msgtype=r900 -decimation=8 as flags to run on embedded devices (such as RPi and others) without losing frames on rtl_tcp. At the moment I'm testing using an embedded x86 Linux.

If I can provide any more information or logs to help debug this, please let me know.

"bad" version: 
Build Date: 01-28-2016
Commit:     v0.6.2-28-ge8ea9d9

"good" version: the -version flag doesn't work :) but its MD5 is 96fe5cd30decf40cb50be74fe699d20f and I got it (again) today from rtlamr_linux_386.tar.gz (MD5 17716b7a1c15535f779f1fc12bdaf294) listed for 0.6.2 "releases".

bemasher, 2016-03-07:
If you could make a sample capture file using the '-samplefile' flag on the last working version I can take a look at it. Somewhere between 20 and 30 messages should be enough to test with.

leomrlima, 2016-03-07:
Here you are! I left it running for aprox. 1 hour, and many messages were captured.
[rtlamr062.capture.zip](https://github.com/bemasher/rtlamr/files/161915/rtlamr062.capture.zip)

bemasher, 2016-03-08:
What flags did you use to create the capture file?

leomrlima, 2016-03-08:
-msgtype=r900 -decimation=8 -samplefile

closed 2016-03-08 by bemasher

bemasher, 2016-03-08:
I've built a new release for this fix: https://github.com/bemasher/rtlamr/releases/tag/v0.7.1

commit 31bbf3f1 (2016-03-08): Use decimated packet configuration in preamble search. Fixes #47.