bemasher/rtlamr #48: scm+ types?

rpaditya, 2016-03-09:
What do the following scm+ EndpointType's correspond to?

0x00
0x81
0x0B
0x08
0xA9
0x37
0xFA
0xF1
0x82
0x98
0xAA
0x05
0x81
0x81
0x96
0x85
0x2D
0xAE
0xC8
0x2C
0xA7

bemasher, 2016-03-10:
I don't actually have an SCM+ capable meter to test with, but my understanding from what little documentation there is on the SCM+ protocol is that the least significant nibble of the endpoint type corresponds to the commodity type used in the rest of the ERT protocol, so from your values:

```0, 1, 2, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14```

These "should" correspond to meters in the meter list: https://github.com/bemasher/rtlamr/blob/master/meters.csv

rpaditya, 2016-03-10:
thanks -- this is really odd (to me) -- I see a lot of scm+ messages (and nothing for other protos) -- what are the fields in the csv output format?

for example:

2016-03-09T18:35:55.784116742-05:00,0,45056,0x16a3,0x49,0x0,194075282,344578944,0x448d,0x7eeb

2016-03-09T18:42:27.721659352-05:00,0,45056,0x16a3,0x5a,0xb8,1241879281,1536023360,0x5475,0xe62e

bemasher, 2016-03-10:
See the fields listed here: https://github.com/bemasher/rtlamr/blob/31bbf3f14626837d4a724d26db363d8880b6913a/scmplus/scmplus.go#L141-L151

```
Timestamp,Offset,Length,FrameSync,ProtocolID,EndpointType,EndpointID,Consumption,Tamper,PacketCRC
```

You might want to verify that the messages are valid based on seeing duplicate values more than one  message for the `EndpointID` field. It is possible for false positives to decode properly despite being just noise.

rpaditya, 2016-03-10:
I have a couple of days of listening, 200+ messages, each with a unique EndpointID -- odd and suspicious...

bemasher, 2016-03-10:
For a couple of days worth of listening there should be thousands of messages (if multiple meters) rather than hundreds. I suspect that there aren't any SCM+ compatible meters in your area and the decoder is picking up noise that passed the CRC.

closed 2016-03-19 by bemasher