bemasher/rtlamr #49: Memory Leak in rtlamr

tracyj, 2016-03-29:
I created a program that uses rtlamr and logs meter readings along with other stuff at midnight every night.  It runs for close to a month, and then quits working.  I decided to see if my program was releasing all of the memory that it used, and found out that it wasn't.  So then I tracked it down to rtlamr by running it by itself, and executing free after each run.  I am running this on a Raspberry PI 2.  The command that I am using is as follows.  I also downloaded the latest release of rtlamr today.

/home/pi/goplay/bin/rtlamr -quiet=true -single=true -format=json 14233925 14233926

I have attached the output from running rtlamr followed by free until it started failing.
             
[memory_leak.txt](https://github.com/bemasher/rtlamr/files/194405/memory_leak.txt)

tracyj, 2016-03-30:
No.  I start rtl_tcp as a service at boot, and let it continue to run.

raspberrypi:/etc/init.d> more rtltcpd.sh
#!/bin/bash
# /etc/init.d/rtltcpd.sh

### BEGIN INIT INFO
# Provides:          rtl_tcp for automatic meter reading
# Required-Start:    $remote_fs $syslog
# Required-Stop:     $remote_fs $syslog
# Default-Start:     2 3 4 5
# Default-Stop:      0 1 6
# Short-Description: Example initscript
# Description:       This service is used to manage rtl_tcp
### END INIT INFO

case "$1" in
    start)
        echo "Starting rtl_tcp"
        /usr/local/bin/rtl_tcp >/dev/null 2>&1
        ;;
    stop)
        echo "Stopping rtl_tcp"
        killall rtl_tcp
        ;;
    *)
        echo "Usage: /etc/init.d/rtltcpd.sh start|stop"
        exit 1
        ;;
esac

exit 0
raspberrypi:/etc/init.d>

bemasher, 2016-03-30:
Before I get too far into this could you tell me if the memory usage remains the same after killing rtl_tcp once free memory has gotten low from running rtlamr multiple times?

I suspect that rtl_tcp isn't cleaning up it's thread/linked-list appropriately when the worker thread exits, and this may be due to rtlamr not closing the connection politely. It's extremely unlikely that rtlamr itself has a memory leak since go is garbage collected, I have very little control over what stays in memory while it is running and I'm not doing anything unsafe that the garbage collector couldn't handle.

tracyj, 2016-03-31:
I have included a log of running rtlamr multiple times until the memory gets fairly low.  I execute free after every run.  At that point, I find the process for rtl_tcp, and kill it.  I then execute free again, and it looks like all the memory has been returned.

[memory_leak_kill_rtl_tcp.txt](https://github.com/bemasher/rtlamr/files/198762/memory_leak_kill_rtl_tcp.txt)

bemasher, 2016-04-10:
I've confirmed that rtlamr is closing it's connection to rtl_tcp appropriately and the leak exists in rtl_tcp. I'll take a look at filing a bug report for rtl_tcp but for the time being I'm closing this issue.

In the meantime, you'll want to kill and restart rtl_tcp once every several rtlamr executions.

Another option would be to figure out which frequencies your meter operates on and set the center frequency appropriately as well as shortening the symbol length to 8. On most electric meters this will produce a reading once every 25 minutes and require minimal cpu resources.

closed 2016-04-10 by bemasher

tracyj, 2016-04-10:
Thank you for your dedication and hard work to this project.

benschumacher, 2024-01-20:
This issue is ancient, but I fixed a memory leak related to connections closing in `rtl_tcp` in my fork: https://github.com/benschumacher/rtl-sdr/tree/rtltcp_memory_leak

This has been submitted as a PR to the upstream project: https://github.com/pinkavaj/rtl-sdr/pull/5