bemasher/rtlamr #50: calculating average KWHr usage

quarterturn, 2016-04-04:
I've been reading the IR LED of my Itron meter to calculate a WHr average. You get 1 blink per WHr, and from there you can just count for whatever time period you want and display it. Of course, the IR LED is weak, and you have to shield it from sunlight, and you need wires.

From rtlamr, it seems that I get data about once every three minutes. What have you been using to calculate your power usage? I find it most useful as far as energy-consciousness goes to have a display of how much power is being used right now, vs something like the whole day.

bemasher, 2016-04-04:
I'm not sure I know what you mean.

Keep in mind that the frequency of transmissions and the frequency of consumption update may not be the same. For example, in bubble-up mode, most meters will transmit a new message once every 30 seconds, but the actual consumption value may only be updated once every 5 minutes.

quarterturn, 2016-04-04:
If I understand correctly, it sounds like the meter doesn't send data on a guaranteed schedule. So, if I want a moving average, I'll need to count the time between each reading received and use that to calculate the average power usage.

brownleeb, 2016-04-04:
I was in very much the same situation.  I used the pulse count off the IR LED to show my instantaneous usage.  Once a day I would run rtlamr to sync up the meter and the accumulator I stored on my RPi.
I use an arduino to do the 'AD conversion' and then feed the pulse out from the arduino digital pin to the counter on the RPi.
I had to do this because my meter only sent out full kwh numbers, but generally each 24 hours I was within 10-20 watts of being in sync.
My project is at https://github.com/brownleeb/Electric-Meter-Monitor but its a little old and I never finished the project after uploading it.  I tracked usage over the previous 1, 5 and 60 minute periods.

Feel free to contact me if you want to know more....

bemasher, 2016-04-04:
Power meters for the most part will transmit once every 30 seconds on a different channel out of the 50 available. These are spread out throughout the 915 MHz ISM band such that an rtl-sdr can only receive 12-13 adjacent channels simultaneously. The hopping pattern used is some "random" permutation of the 50 channels and usually fixed (the same pattern is used repeatedly by a particular meter).

With rtlamr's default configuration you can expect there to be 30s to 5m between messages since some messages will be transmitted on a frequency outside the sampled bandwidth of your rtl-sdr.

What I meant with the difference in update rate was that you may see duplicate messages, messages with identical consumption values. This doesn't always mean that you consumed no power between the messages, just that the meter hasn't yet provided an updated consumption value to the transmitter.

Another thing to be aware of is that the consumption value found in SCM messages is usually measured in 1/100th's of a kWh and is cumulative, to determine usage over time you'll need to look at the difference between consecutive messages. If your meter supports IDM messages, this is easier since the intervals each represent the difference in consumed power over 5 minute intervals. Each IDM message contains 47 intervals, which covers ~3.9 hours worth of power usage.

If you decide to only use SCM messages, knowing when the message was received is important to calculating the power consumed since the last message.

quarterturn, 2016-04-05:
Thanks for the reply. Good to know there are 'missing' messages due to rtl-sdr limitations. Looks like I need to find a good non-blocking method of catching the output of rtlamr in python so I can count the time between messages, in order to get a 'live' watthour consumption readout.

brownleeb, 2016-04-05:
I'll post my code to my github for Electric Meter Monitoring.  All I do is start rtl-tcp and rtlamr with output to a file.  Then I have a php program that reads the output file.  Dont know why I used php over python, just what I did at the time.
As I said, my older meter only reported kwh, but many newer meters around me did 1/1000ths.  My code would read the usage value until it changed, sync my accumulator and then shut everything down.  I did the sync at night so something it would take as much as 2-3 hours to notice the change.
You dont indicate which value your meter is reporting.  
The whole RTL package chews up a good bit of cpu, but it doesnt hurt if its an RPi or something that isnt really doing anything else anyway.  Also, once you zero in, you can put some command line args to make it more efficient.

bemasher, 2016-04-05:
@quarterturn: You can use the timestamps rtlamr records when using json output:

File: listen.py
```python
#!/usr/bin/env python

import sys, json
import dateutil.parser

while True:
	line = sys.stdin.readline()
	if not line:
		break
	msg = json.loads(line.strip())
	timestamp = dateutil.parser.parse(msg["Time"])
	print timestamp, msg["Message"]["Consumption"]
```

```
$ rtlamr -quiet -format=json | python -u listen.py
2016-04-05 17:39:33.431087-06:00 4708199
2016-04-05 17:39:33.935947-06:00 3933568
2016-04-05 17:39:34.541132-06:00 1689792
2016-04-05 17:39:35.153327-06:00 630070
2016-04-05 17:39:35.267723-06:00 2768950
2016-04-05 17:39:35.710518-06:00 908977
```
This is just a basic example but you can build from it.

@brownleeb: On an RPi2 rtlamr can run without sample loss up to about -symbollength=31 which is about half the default bandwidth.

closed 2016-04-10 by bemasher