bemasher/rtlamr #52: r900 working?

ryanbagwell, 2016-04-18:
Just downloaded this and fired it up. Am able to read SCM data, but r900 doesn't seem right. All IDs and consumption figures are 0. Here's the output:

````
$ rtlamr -msgtype=r900
21:20:29.552553 decode.go:82: CenterFreq: 912380000
21:20:29.552749 decode.go:83: SampleRate: 2359296
21:20:29.552762 decode.go:84: DataRate: 32768
21:20:29.552773 decode.go:85: SymbolLength: 72
21:20:29.552783 decode.go:86: PreambleSymbols: 32
21:20:29.552793 decode.go:87: PreambleLength: 4608
21:20:29.552803 decode.go:88: PacketSymbols: 116
21:20:29.552813 decode.go:89: PacketLength: 16704
21:20:29.552823 decode.go:90: Preamble: 00000000000000001110010101100100
21:20:29.552851 main.go:93: GainCount: 29
{Time:2016-04-17T21:20:31.520 R900:{ID:         0 Unkn1:0x00 NoUse: 0 BackFlow:0 Consumption:       0 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2016-04-17T21:20:32.284 R900:{ID:         0 Unkn1:0x00 NoUse: 0 BackFlow:0 Consumption:       0 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2016-04-17T21:20:32.732 R900:{ID:         0 Unkn1:0x00 NoUse: 0 BackFlow:0 Consumption:       0 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2016-04-17T21:20:33.180 R900:{ID:         0 Unkn1:0x00 NoUse: 0 BackFlow:0 Consumption:       0 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2016-04-17T21:20:37.844 R900:{ID:         0 Unkn1:0x00 NoUse: 0 BackFlow:0 Consumption:       0 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2016-04-17T21:20:38.522 R900:{ID:         0 Unkn1:0x00 NoUse: 0 BackFlow:0 Consumption:       0 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2016-04-17T21:20:38.635 R900:{ID:         0 Unkn1:0x00 NoUse: 0 BackFlow:0 Consumption:       0 Unkn3:0x00 Leak: 0 LeakNow:0}}
````

Am I doing something wrong?

bemasher, 2016-04-18:
Yeah, turns out that's a side-effect of trimming all of the intermediate buffers (see faba1096). I'll look at making internal buffer length configurable based on the protocol in use.

In the meantime, anything at or before [v0.7.1](https://github.com/bemasher/rtlamr/releases/tag/v0.7.1) should work for R900 meters.

bemasher, 2016-04-18:
If you built rtlamr from source, could you test this fix? 082d45238820fd9092a92a0d60415dca145a7504

If not, I can build a release for your os/arch to test with.

ryanbagwell, 2016-04-18:
Yes, will try to test later today. Thanks for jumping on this.

ryanbagwell, 2016-04-18:
Looks like it's working:

````{Time:2016-04-18T08:19:08.968 R900:{ID:1541531110 Unkn1:0xA3 NoUse:33 BackFlow:0 Consumption:  154039 Unkn3:0x00 Leak: 0 LeakNow:0}}````

Still some issues with decoding the Unkn1 and Unkn3 bits, but the important data is coming through. Thanks again!

bemasher, 2016-04-19:
`Unkn1` and `Unkn3` fields are possibly meter-specific, whatever happens to be transmitted is still shown in those fields but their meaning is left as an exercise to the user.

Thanks for catching the problem, I don't have an R900 compatible meter to test with.

closed 2016-04-19 by bemasher

gbdavidx, 2022-07-04:
what parameter can i use to change the channcel frequency it uses?