bemasher/rtlamr #54: Error in logfile when set to run on startup

hansthompson, 2016-05-02:
Hi bemasher.  When I set rtl_tcp to run on startup in  /etc/rc.local and set up rtlamr to run ```@reboot``` with crontab I end up with this erroro in my log file for rtlamr.  

```15:38:21.897423 recv.go:55: Error connecting to spectrum server: dial tcp 127.0.0.1:1234: connection refused```

I can confirm that rtl_tcp is running because rtlamr runs when I execute from the command line.  I'm not sure if this is a bug or my own ignorance.

bemasher, 2016-05-02:
It takes a non-zero amount of time for rtl_tcp to search for and initialize a dongle for use before it can be connected to, you'll need some kind of delay before trying to start rtlamr on startup.

hansthompson, 2016-05-04:
Ah. Yes. I gave it a little time and it works.

closed 2016-05-04 by bemasher