bemasher/rtlamr #55: Support for Landis+Gyr AMI Focus Meters

njreid, 2016-05-06:
Hi bemasher,

Thanks very much for all your work on this library and utility! Seems to be working well for my Philadelphia Gas Works ERT meter, using the scm message type. 

It looks like PECO (Electric Utility) uses the Focus AXR-SD AMI meters - do you have any ideas about how to start reverse engineering that protocol? Given it's an encrypted mesh, presumably that makes it harder to snoop? Does it need to be actively queried? Are there any technical resources about the protocol?

Cheers,

Nic

njreid, 2016-05-06:
I found this overview of the Landis+Gyr system, which is pretty detailed. 
http://nepa-ru.com/Landys+Gyr_files/aim/01_web_ami_description_en.pdf

bemasher, 2016-05-06:
Based on prior research, support for AXR-SD meters is probably never going to happen because of two things: it is possibly encrypted and it uses an unsupported modulation (FSK).

I'm not willing to support or attempt to reverse engineer encrypted communications systems like your meter. However, my experience is that even though an AMI system supports encryption that does not necessarily mean it is configured/used. You're welcome to poke around and take a look at the messages your meter is transmitting but it's a really grey area leaning more towards the black side of things.

Implementing FSK requires that you know the frequencies the meter will transmit on ahead of time and the order in which they are used in the case of FHSS. Implementing a promiscuous FSK demodulator that can handle many channels simultaneously requires a lot more CPU resources than most of the single-board computers this software is intended for can provide.

closed 2016-05-06 by bemasher

rrauenza, 2017-03-08:
I was confused looking at https://github.com/bemasher/rtlamr/blob/master/meters.csv that it was supported (so bought a dongle!) -- do you want to add a column to that CSV to show that this meter is unsupported?

bemasher, 2017-03-09:
You need to pay attention to the FCC ID of your meter rather than the meter itself. Meters in general are frequently available in multiple configurations that support different protocols and radios.

bemasher, 2021-05-27:
For the curious among you, this page has quite a lot of good information regarding Landis+Gyr meters and decoding the protocol at least some of them use.

https://wiki.recessim.com/view/Gr-smart_meters_Setup_Guide