bemasher/rtlamr #56: Capturing ERT packet for peak 15 min. load?

hansthompson, 2016-05-10:
In my area, the meters are broadcast a 15 minute peak load in addition to a SCM message of cumulative consumption.   

Is there a specific message type that would broadcast this 15 minute load? If it is coming off the same Intron meter for the electrical SCM message is it broadcasting in SCM as well?

bemasher, 2016-05-10:
What documentation do you have that supports this? The only other kind of message I'm aware of that power meters might transmit are IDM messages which include differential interval data at 5 minutes per interval.

hansthompson, 2016-05-18:
This is what I've heard from the director of IT from my local utility.  I'll check the IDM message.

closed 2016-10-02 by bemasher