bemasher/rtlamr #59: High CPU utilization - r900 Parser is called when zero preambles found in Message?

kb1ca, 2016-06-11:
Listening for r900 packets on my Pi 3 uses 100% of one CPU core, regardless of the number of meters audible to the receiver or the message receive rate. Poking around in the code it seems that Parse() is called even when there were zero preambles found in the Message, causing p.Filter() and p.Quantize() to do unnecessary work.

Checking for zero preamble indices and immediately exiting the parser function reduces the average cpu time to between 50% and 60%, without any obvious loss of messages. This also eliminates the stream of `ll+, now...` messages usually emitted by rtl_tcp for the first minute or two of operation. 

Interestingly inserting the return before the cfg/copy at the start of the function causes no preambles to ever be found. Unfortunately I have little experience with Go, so delving into why that is is beyond my ability.

The exact patch is: 

func (p Parser) Parse(indices []int) (msgs []parse.Message) {

        cfg := p.Decoder.DecCfg
        copy(p.signal, p.signal[cfg.BlockSize:])
        copy(p.signal[cfg.PacketLength:], p.Decoder.Signal[cfg.SymbolLength2:])

        // Why are we called with no preambles?
        if len(indices) == 0 {
                return
        }

        p.Filter()
        p.Quantize()

bemasher, 2016-06-11:
Packets will span multiple sample blocks, if you skip calling `p.Filter()` and `p.Quantize()` when you don't detect a preamble you lose signal history required when you do later detect the preamble.

The copies at the beginning of the Parse method rotates the signal buffer and writes new samples to the beginning of the buffer. Returning before these copies means that when you don't have any detected preambles you will lose signal history as above.

The R900 decoder is heavier on cpu usage because the preamble is Manchester coded while the rest of the packet uses three unique 4-chip symbols that each require their own matched filter to decode which requires both `p.Filter()` and `p.Quantize()` to determine bit values at each sample in the signal.

closed 2016-06-11 by bemasher

kb1ca, 2016-06-13:
Thanks for the clarification.