bemasher/rtlamr #60: flexnet meters?

amitgandhinz, 2016-07-11:
Hi

Thanks for this great software.  

I was able to pull numbers from the gas meter, but i can't detect the Electric meter (FlexNet 530X), or the Water Meter (Neptune).

Has anyone been able to pull those numbers, either with this software or any other repo?  Im guessing these need to be "woken up" to transmit data, but not sure how to do that.

Thanks

bemasher, 2016-07-11:
Neptune meters to my knowledge are all designed to transmit at regular intervals, they don't require a wake-up signal. Try `rtlamr -msgtype=r900` to listen to your Neptune meter.

FlexNet is an unsupported protocol.

amitgandhinz, 2016-08-04:
I still haven't been able to pull data from the Neptune Water Meter.  

Interestingly, I don't see a transmitter of any kind on the water meter leaving me wondering if it is indeed smart.  I would be surprised if it isn't since I live in a new construction community in GA (< 1 year old).

bemasher, 2016-08-04:
You may need to contact your utility to find out what technology is used to read the meters (if any). Let me know if you find anything out about it.

closed 2016-09-11 by bemasher

happyleavesaoc, 2016-10-13:
Has anyone been able to monitor a FlexNet 530X via any method? Are there any plans to do so?

bemasher, 2016-10-13:
There is little or no documentation about Sensus' RF protocol for FlexNet meters and the only information I've been able to find about it suggests that it is encrypted. I neither have access to a meter that supports this protocol nor cooperation from a utility which has provisioned one that would be willing to provide the appropriate cryptographic key to decrypt received transmissions and read the meter.

In general utilities aren't shy about allowing their customers to read their meter if they have one of the various consumer products that can receive the protocol and are willing to provide a particular customer the appropriate key to read their meter.

Unfortunately I'm not one of those customers and it's usually not easy to obtain a meter for testing, particularly if you want one that is already provisioned so that it will transmit and so that I can know the key used to decrypt it's transmissions.

In short, very probably not going to happen.

chaplijt, 2017-09-27:
Just moved so I have one of these Flexnet 530x meters. The elec co, seems to be able to read directly from their office . Discussion with customer service indicates that the meters communicate using a "cell tower". maybe 3g communication?

Their web site allow me to download usage data . It is not  exactly hourly, but quantized on even kwh useage (with a minimum 1 hr resolution.  trying to make sense how to use the csv  file for my solar analysis

dougalcampbell, 2018-01-30:
Bummer. I've got a Flexnet 530x, too. I was picking up data from scm meters in my area, but I couldn't find *anything* that correlated with my kWh. So then I searched and found this thread.

nitingautam, 2018-07-10:
Any updates on Sensus FlexNet meters data read?

billatq, 2020-01-28:
Looks like it depends. The Sensus water meters in my area are 510Ms, and it looks like depending on the age of them, it's similar in that it sends out a narrowband burst of data.

jasonshearer, 2021-09-06:
> Hi
>
> Thanks for this great software.
>
> I was able to pull numbers from the gas meter, but i can't detect the Electric meter (FlexNet 530X), or the Water Meter (Neptune).
>
> Has anyone been able to pull [...]

What model gas meter do you have? Curious if it is a Sensus Flexnet. Trying to get my usage data.

kellcomnet2, 2021-10-02:
I have a sensus Flexnet 100GM-A gas meter
[image]
I am willing to open ssh access to a rPi with a https://www.adafruit.com/product/1497 hooked up to it from a specific origin ip if it would help. I have 15 minute water and electric readings already from my utilities but gas is all monthly reads and I would like to be able to do monitoring on it as well.

XtremeOwnageDotCom, 2023-01-18:
I happen to have one of these meters: https://cdn2.webdamdb.com/md_QGeBOZCyWrM0.jpg.pdf?v=3

FCC Certification is here: https://fcc.report/FCC-ID/SDBIDTB005/4095810.pdf

Appears, flexnet uses 900-960mhz. Some models, apparently also supports zigbee, and other methods too.

Will be doing a bit of SDR capture in a few days when a secondary sdr modules arrives, if the data helps, can post it here.

kalbasit, 2023-06-24:
@XtremeOwnageDotCom did you manage to put something together? I have the Sensus 520R installed by the water company.

XtremeOwnageDotCom, 2023-06-24:
> @XtremeOwnageDotCom did you manage to put something together? I have the Sensus 520R installed by the water company.

After pointing rtll433 to 915mhz, it actually pulled the data in automatically.. for my meter.

Assuming you have 433mhz devices too, you can either use a dual antenna set to hop frequency every now and then... or use two tuners, one for 433, one for 915

billatq, 2023-06-24:
@XtremeOwnageDotCom Interesting. How often does it chirp, and how does it display on `rtl_r33`?

I did notice that when I run `rtl_433 -f 915000000` I can find my gas and electric meter, but I'm not seeing the Sensus 510M. It's possible that mine is set up to only transmit when someone comes by to collect the data though.

XtremeOwnageDotCom, 2023-06-24:
In the case of the meter I was testing with-

If memory serves me- it would send data every 10 minutes, followed by every
2.5m. A very unusual.... schedule.

Ie-

8am, 8:10, 8:12, 8:22, 8:24, etc.

kalbasit, 2023-06-27:
@XtremeOwnageDotCom how did you run rtl_433? I'm using [this RTL-SDR](https://www.amazon.com/dp/B01GDN1T4S) with its small ~11.7cm (4.6in) antenna and running it like this 

```
❯ rtl_433 -f 915e6
rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/yl/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 191 out of 223 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-223 ]
Found Rafael Micro R820T/2 tuner
Exact sample rate is: 1000000.026491 Hz
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 915.000MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 1000000 Hz at cutoff 200000 Hz, 5.0 us
```

It's been running 2.5h and it printed nothing so far.

XtremeOwnageDotCom, 2023-06-27:
Your setup appears correct. There is a chance your particular meter isn't
915m.

Check the FCC ID number listed on the back of it, to see if you can
determine what frequency it broadcasts on.

kalbasit, 2023-06-27:
[image]
That's the one I have (minus the ID, blurred).

kalbasit, 2023-06-27:
Looks like it 956M from the top-right corner.

kalbasit, 2023-06-28:
No dice, unfortunately. It's been running for nearly a day and hasn't printed anything. I swapped the antenna with the 17cm one halfway through to confirm it's not an antenna size problem.

pir8radio, 2024-03-08:
The battery devices often only TX once every hour or more.   They collect data in X min increments, buffer, then send their data. Depending on the customers (utility) battery life want/need.    Some customers only read their meters once a day, or once a week and that internal battery will last more than 10 years.   They are FSK-13 modified.  So its supposedly an odd protocol.   This is an old document from when they were fsk-7 but still same theory.     https://sensus.com/wp-content/themes/sensus/inc/webdam_download.php?asset_id=17563312

pir8radio, 2024-03-08:
https://fccid.io/SDBFLEXELS/User-Manual/Manual-1621763