bemasher/rtlamr #62: Support for Itron C1SX residential models?

heresjonny1, 2016-07-27:
Hello,
At my residence they are using Itron C1SX. I would like to try and receive my smart meter's information, unfortunately I couldn't get your program to work. Upon further reading it appears that it uses RS300 protocol:

https://web.archive.org/web/20140930034923/https://www.itron.com/na/productsAndServices/Pages/CENTRON.aspx

bemasher, 2016-07-27:
It doesn't use either the "R300" or R900 protocol. It supports both SCM and IDM messages as well as Itron's proprietary fixed network protocol. In general fixed network protocols are encrypted and I'm not prepared to reverse-engineer or support such a protocol.

Before giving up on this entirely, if you can provide me with the FCC ID for your particular meter and try both SCM and IDM message types, that'll help in figuring out what frequencies your meter operates on and how your meter might be provisioned. From documentation it appears that your meter should operate between 910 MHz and 920 MHz. Give both of these commands a try:

```
rtlamr -msgtype=scm -centerfreq=915M
```

```
rtlamr -msgtype=idm -centerfreq=915M
```

heresjonny1, 2016-07-28:
Unfortunately, neither command worked. I do see it transmitting from 915Mhz to about 920Mhz using an analyzer. Also there appears to be no FCC ID on the device itself.

Photos: http://imgur.com/a/Pk9Qg

I did manage to find to very similar units and their ID's:

https://fccid.io/F9CC1C-1 (C1SC)
http://www.ic.gc.ca/pics/lm/electric/ae/0920r31.pdf (This one even mentions C1SX)

bemasher, 2016-07-29:
Unfortunately your meter does not communicate wirelessly. Your meter communicates with the utility via the power line.

http://www.cooperindustries.com/content/dam/public/powersystems/resources/library/1110_AMR_AMI/B110008150.pdf

closed 2016-07-29 by bemasher