bemasher/rtlamr #63: Adjustable data rate

tescott, 2016-09-04:
I've got a Schlumberger R900, FCC ID F9CTALSRFW1 water meter.  This thing appears to be following the R900 protocol shown on http://bemasher.net/rtlamr/protocol.html when I decode its transmission manually, but it is transmitting at a data rate of 16384 rather than 32768.  Is there any possibility of making rtlamr's data rate an adjustable parameter?

bemasher, 2016-09-04:
You can specify both symbol length and sample rate, together these will set the data  rate.
SampleRate = SymbolLength * DataRate

rtlamr -symbollength=72 -samplerate=1179648

The above command should give a data rate of 16384. Keep in mind that the R900 protocol's preamble is manchester coded while the rest of each message is encoded with at set of 3x unique symbols made up of 4x chips each and their negations.

Would you be willing to send along a signal sample from your meter?

tescott, 2016-09-04:
Thanks for the tip!  I'll retry things with that.  See attached for a SDR#
recording to a .wav  I recorded things at 2.4Msps.  If another format is
better, let me know.

Where can I find further information about the R900 protocol?  I can see
the pre-amble quite clearly.  The remainder of the message is a mystery to
me at this point.

Thanks again!

tescott, 2016-09-04:
Erm.  Doesn't look like Github issues retain attachments.  Here's a (temporary) Dropbox link: https://www.dropbox.com/s/8f1udp7ogq6nuz6/schlumberger_sample_F9CTALSRFW1.wav?dl=0

bemasher, 2016-09-04:
If you could use this command I could more easily analyze it in rtlamr's test fixtures:

```rtl_sdr -f 915080000 -s 2359296 -g 0 -n 13107201 r900.bin```

Should produce a 25MB file centered on the spectrum where your meter should be transmitting.

tescott, 2016-09-04:
Dropbox link here: https://www.dropbox.com/s/0ae6nmbupa5j9ix/r900.bin?dl=0

tescott@msi-vbox ~/projects/rtl-sdr-keenard/build/src $ ./rtl_sdr -f 915080000 -s 2359296 -g 0 -n 13107201 r900.bin
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Sampling at 2359296 S/s.
Tuned to 915080000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...

User cancel, exiting...

bemasher, 2016-09-04:
It's transmitting at the appropriate data-rate for rtlamr but is failing because it's transmitting malformed R900 symbols. If you could make a few more samples like the one above I can dig a little deeper.

tescott, 2016-09-04:
How are you coming up with the data rate?  I took a recording using SDR#.  I brought the .WAV recording into octave and looked at the # of samples between two of the pre-amble '0' bits and saw there were roughly 150 samples.  At a sample rate of 2.4Msps I get 150/2400000 = 62.5us -> ~16kbps.  Just want to know where my calculations are going wrong here.

Currently collecting more data for you to check out.  

A couple of questions:

1) What are you using to analyze the .bin files?

2) The unit transmits using frequency hopping across 50 channels at a 4s rate.  How is it that 915.08MHz is able to pick anything up in such a short amount of time?  I mean, unless I was extremely lucky, I was thinking the trace you had me perform wouldn't have contained any data.

Thanks.

tescott, 2016-09-04:
More sample data: https://www.dropbox.com/s/d9x2kqauht9e07k/r900_moredata.zip?dl=0

bemasher, 2016-09-05:
You're right about the data rate based on how you measured it. What rtlamr displays when logging is a side-effect of a convention I used really early on in the project where data rate was based on the smallest component of the signal, a chip. So it says the data rate is 32768 because everything is calculated from the length of a single chip, but because the signal is manchester coded, each symbol is two chips thus 16384.

I'm using audacity's raw data import feature to view sample files.

That's actually frequent enough to usually catch one in the short time `rtl_sdr` is running. The meter transmits on one of 50 channels between 911.08 MHz and 919.08 MHz and we're listening to ~2.4 MHz worth of spectrum centered at 915.08 MHz which will catch ~30% of the channels it might transmit on. It's also possible that your meter isn't the only one transmitting nearby.

tescott, 2016-09-05:
Thanks for the clarification!  I think the main trick now is to figure out how the data payload is encoded.  I'll be recording my water levels and RF readings over the next few days to see if I can figure out what it's doing.  

FWIW, here's my analysis flow:

SDR# for recording -> Audacity for trimming and viewing -> Octave for data manipulation.

tescott, 2016-09-12:
FYI-- I'm starting to make sense out of the data payload.  I figure another week or two and hopefully I will have the encoding understood.  Hopefully.

tescott, 2016-09-16:
I just received a new Neptune R900 meter from the city.  Tried using rtlamr with it but that doesn't seem to read its transmission.  The raw preamble looks the same but rtlamr apparently can't decode the payload.

I generated the following example using "rtl_sdr -f 915080000 -s 2359296 -g 0 -n 13107201 r900_new.bin" if you have the time to take a look:

https://www.dropbox.com/s/xx780nd0yk1gxlo/r900_new.bin?dl=0

I'm guessing it uses the same broadcast format / protocol that my old meter used.

tescott, 2016-09-17:
Success!  At bemasher's request, I reverted to 0.8.0 and that worked.  Closing the issue.

closed 2016-09-17 by tescott

bemasher, 2016-09-17:
Fixed in a0ec0e824206cfa6d73b6f6e99910544375cadb5.