bemasher/rtlamr #64: Rolled-over electric meter not decoded

xpectdelays, 2016-09-10:
This past week, our Schlumberger Watt-Hour meter rolled over to 100,000 kWh. The digital display outside now says something like 00180 kWh. Rtlamr had no trouble reading the meter (last successful reading was only a few days prior at 99,800 something kWh), now, it doesn't show up at all. This is a fairly populated area so there is quite a bit of chatter, like before, except that our meter doesn't show. I know the serial number as it is printed on the meter face and have successfully used it with the 'filterid' option. I have tried adding diagnostic 'println' to decode.go, instead of silently 'bailing', but have seen no clues as to the cause, i.e. short packet or bad checksum. This is a meter installed by National Grid/Niagara Mohawk, it's probably 10 years old if not older.

Some other resources online regarding the ERT protocol indicate variable-length packets are at least possible. Could it be that these are not being decoded properly?

bemasher, 2016-09-11:
Both SCM and IDM are fixed length packets. I find it highly unlikely that the consumption register rolling over would cause it to switch to transmitting a different message type. You'll probably want to contact your utility and let them know that it rolled over and you're having trouble reading it.

xpectdelays, 2016-09-11:
I don't think the message format has changed, more likely, it's either subtly different from Itron's own implementation or there's an issue in rtlamr where it happened to work until now. I suppose I will try again at debugging, however, it's difficult given that there are probably a dozen or more meters in range and that I have very little experience with Go.

I have a hard time imagining that the meter has a problem. This county is home to half a million people and these are the only electric meters I've seen anywhere around here. If they had a rollover issue, it'd have been found and addressed _years_ ago.

But stranger things have happened. Guess I'll find out!

xpectdelays, 2016-09-11:
And just like that, it appears to be working again. The value shown in rtlamr is just '183' which is what the display outside says (00183). I can't even begin to speculate what changed from ... well, three hours ago, because I can't think of anything that makes sense. It's been a problem since at least Friday afternoon. Hopefully it stays working.

Thanks for putting up with my goofiness and thanks for this program. It will be nice to be able to read the meter in a few months when it's snowy and cold.

closed 2016-09-11 by xpectdelays