bemasher/rtlamr #67: Network Interval Message (NIM) Protocol

cniesen, 2016-11-25:
I have the 100W water meter (model 100WRD).  The rtlamr successfully reads the consumption value via the scm+ message type.  I also get the reading from a neighboring meter.

Apparently the water meter also supports the Network Interval Message (NIM) protocol which includes additional meter data.  The protocol seems to be well documented in the attached document.  Have you investigated that protocol and would it be possible to add it?

[Water AMI 100W Network Interval Message.pdf](https://github.com/bemasher/rtlamr/files/612914/Water.AMI.100W.Network.Interval.Message.pdf)

bemasher, 2016-11-29:
I've written a prototype NIM decoder but have never had any data from a meter in the wild to test with. Adding it is relatively straight forward since I've been able to write a frequency demodulator for rtlamr. There is some extra logic necessary to determine how long the packet will be and adjust accordingly while listening that I've not yet written.

If you would be willing to provide sample data I can probably have something up and running without too much trouble assuming your meter is configured to transmit NIM's.

cniesen, 2016-11-29:
Sure, do you have instructions on how to capture the data?

bemasher, 2016-11-29:
Did you download a binary release of rtlamr or did you build from source? I wrote a tool that implements a simple squelch to capture only "interesting" signals in what should otherwise be noise.

cniesen, 2016-11-29:
I did build it with the `go get github.com/bemasher/rtlamr` command

bemasher, 2016-11-29:
Please see this project for details: https://github.com/bemasher/rtlcap

In particular this section: https://github.com/bemasher/rtlcap#squelch

Please record a capture with the following flags:
```-centerfreq=908M -samplerate=2359296 -tunergainmode=false -gainbyindex=0 -bytes=10M```
and an appropriate squelch level set with `-squelch`. If `-gainbyindex=0` doesn't pick up sample blocks fast enough, bump it up in increments of 5, you'll need to figure out a new squelch level whenever you change the gain.

cniesen, 2016-11-29:
Is this capture good?

[sample.zip](https://github.com/bemasher/rtlamr/files/618165/sample.zip)

> C:\Go\workspace\rtlcap\bin>rtlcap -centerfreq=908M -samplerate=2359296 -tunergainmode=false -gainbyindex=0 -bytes=10M -squelch=0.68 -o sample
> 23:02:04.069960 Min: 0.000 Max: 0.682
> 23:02:05.070017 Min: 0.671 Max: 0.682
> 23:02:06.070074 Min: 0.670 Max: 0.682
> 23:02:07.070131 Min: 0.671 Max: 0.681
> 23:02:08.070189 Min: 0.668 Max: 0.682
> 23:02:09.069246 Min: 0.670 Max: 0.682
> 23:02:10.069303 Min: 0.670 Max: 0.682
> 23:02:11.069360 Min: 0.669 Max: 0.681
> 23:02:12.069417 Min: 0.669 Max: 0.682
> 23:02:13.069474 Min: 0.669 Max: 0.682
> 23:02:14.069532 Min: 0.670 Max: 0.682
> 23:02:15.069589 Min: 0.670 Max: 0.682
> 23:02:16.069646 Min: 0.670 Max: 0.682
> 23:02:17.069703 Min: 0.670 Max: 0.681
> 23:02:18.069760 Min: 0.671 Max: 0.681
> 23:02:19.069818 Min: 0.670 Max: 0.682
> 23:02:20.069875 Min: 0.668 Max: 0.682
> 23:02:21.069932 Min: 0.670 Max: 0.682
> 23:02:22.069989 Min: 0.670 Max: 0.682
> 23:02:23.070046 Min: 0.669 Max: 0.683
> 23:02:24.070104 Min: 0.671 Max: 0.681
> 23:02:25.070161 Min: 0.670 Max: 0.683
> 23:02:26.070218 Min: 0.671 Max: 0.681
> 23:02:27.070275 Min: 0.669 Max: 0.683
> 23:02:28.070332 Min: 0.670 Max: 0.681
> 23:02:29.070390 Min: 0.669 Max: 0.681
> 23:02:30.070447 Min: 0.670 Max: 0.682
> 23:02:31.070504 Min: 0.670 Max: 0.683
> 23:02:32.069561 Min: 0.670 Max: 0.682
> 23:02:33.070618 Min: 0.669 Max: 0.682
> 23:02:34.069676 Min: 0.671 Max: 0.681
> 23:02:35.069733 Min: 0.670 Max: 0.682
> 23:02:36.070790 Min: 0.669 Max: 0.684
> 23:02:37.069847 Min: 0.670 Max: 0.681
> 23:02:38.069904 Min: 0.670 Max: 0.683
> 23:02:39.069962 Min: 0.670 Max: 0.682
> 23:02:40.070019 Min: 0.670 Max: 0.683
> 23:02:41.070076 Min: 0.670 Max: 0.682
> 23:02:42.070133 Min: 0.669 Max: 0.683
> 23:02:43.070190 Min: 0.670 Max: 0.682
> 23:02:44.070248 Min: 0.669 Max: 0.681
> 23:02:45.070305 Min: 0.669 Max: 0.681
> 23:02:46.070362 Min: 0.668 Max: 0.681
> 23:02:47.070419 Min: 0.671 Max: 0.682
> 23:02:48.070476 Min: 0.670 Max: 0.681
> 23:02:49.070534 Min: 0.670 Max: 0.683
> 23:02:50.070591 Min: 0.670 Max: 0.681
> 23:02:51.070648 Min: 0.670 Max: 0.683
> 23:02:52.070705 Min: 0.668 Max: 0.681
> 23:02:53.070762 Min: 0.670 Max: 0.681
> 23:02:54.070820 Min: 0.669 Max: 0.682
> 23:02:55.070877 Min: 0.670 Max: 0.682
> 23:02:56.070934 Min: 0.669 Max: 0.681
> 23:02:57.070991 Min: 0.669 Max: 0.682
> 23:02:58.070048 Min: 0.670 Max: 0.682
> 23:02:59.070106 Min: 0.670 Max: 0.681
> 23:03:00.070163 Min: 0.670 Max: 0.682
> 23:03:01.070220 Min: 0.670 Max: 0.682
> 23:03:02.071277 Min: 0.667 Max: 0.683
> 23:03:03.070334 Min: 0.670 Max: 0.682
> 23:03:04.070392 Min: 0.670 Max: 0.682
> 23:03:05.070449 Min: 0.670 Max: 0.681
> 23:03:06.070506 Min: 0.668 Max: 0.682
> 23:03:07.070563 Min: 0.669 Max: 0.681
> 23:03:08.070620 Min: 0.670 Max: 0.682
> 23:03:09.070678 Min: 0.670 Max: 0.681
> 23:03:10.070735 Min: 0.669 Max: 0.682
> 23:03:11.070792 Min: 0.670 Max: 0.683
> 23:03:12.070849 Min: 0.670 Max: 0.682
> 23:03:13.070906 Min: 0.670 Max: 0.683
> 23:03:14.070964 Min: 0.670 Max: 0.682
> 23:03:15.071021 Min: 0.669 Max: 0.682
> 23:03:16.071078 Min: 0.670 Max: 0.683
> 23:03:17.071135 Min: 0.671 Max: 0.683
> 23:03:18.071192 Min: 0.670 Max: 0.681
> 23:03:19.071250 Min: 0.670 Max: 0.682
> 23:03:20.071307 Min: 0.671 Max: 0.682
> 23:03:21.071364 Min: 0.670 Max: 0.682
> 23:03:22.071421 Min: 0.670 Max: 0.681
> 23:03:23.071478 Min: 0.670 Max: 0.682
> 23:03:24.070535 Min: 0.670 Max: 0.682
> 23:03:25.070593 Min: 0.670 Max: 0.682
> 23:03:26.070650 Min: 0.668 Max: 0.682
> 23:03:27.070707 Min: 0.669 Max: 0.682
> 23:03:28.070764 Min: 0.670 Max: 0.681
> 23:03:29.070821 Min: 0.670 Max: 0.682
> 23:03:30.070879 Min: 0.671 Max: 0.682
> 23:03:31.070936 Min: 0.669 Max: 0.681
> 23:03:32.070993 Min: 0.670 Max: 0.683
> 23:03:33.071050 Min: 0.669 Max: 0.680
> 23:03:34.071107 Min: 0.670 Max: 0.682
> 23:03:35.071165 Min: 0.669 Max: 0.684
> 23:03:36.071222 Min: 0.670 Max: 0.681
> 23:03:37.071279 Min: 0.671 Max: 0.683
> 23:03:38.071336 Min: 0.668 Max: 0.682
> 23:03:39.071393 Min: 0.669 Max: 0.682
> 23:03:40.071451 Min: 0.671 Max: 0.681
> 23:03:41.071508 Min: 0.671 Max: 0.682
> 23:03:42.071565 Min: 0.671 Max: 0.682
> 23:03:43.071622 Min: 0.670 Max: 0.681
> 23:03:44.071679 Min: 0.670 Max: 0.682
> 23:03:45.071737 Min: 0.671 Max: 0.683
> 23:03:46.070794 Min: 0.670 Max: 0.681
> 23:03:47.071851 Min: 0.670 Max: 0.682
> 23:03:48.071908 Min: 0.670 Max: 0.681
> 23:03:49.070965 Min: 0.670 Max: 0.681
> 23:03:50.072023 Min: 0.669 Max: 0.681
> 23:03:51.071080 Min: 0.670 Max: 0.682
> 23:03:52.071137 Min: 0.669 Max: 0.682
> 23:03:53.071194 Min: 0.670 Max: 0.682
> 23:03:54.071251 Min: 0.670 Max: 0.682
> 23:03:55.071309 Min: 0.670 Max: 0.682
> 23:03:56.071366 Min: 0.670 Max: 0.681
> 23:03:57.071423 Min: 0.669 Max: 0.682
> 23:03:58.071480 Min: 0.670 Max: 0.682
> 23:03:59.071537 Min: 0.670 Max: 0.682
> 23:04:00.071595 Min: 0.670 Max: 0.682
> 23:04:01.071652 Min: 0.670 Max: 0.682
> 23:04:02.071709 Min: 0.671 Max: 0.683
> 23:04:03.071766 Min: 0.670 Max: 0.682
> 23:04:04.071823 Min: 0.669 Max: 0.682
> 23:04:05.071881 Min: 0.670 Max: 0.683
> 23:04:06.071938 Min: 0.670 Max: 0.681
> 23:04:07.071995 Min: 0.668 Max: 0.681
> 23:04:08.072052 Min: 0.669 Max: 0.683
> 23:04:09.072109 Min: 0.670 Max: 0.682
> 23:04:10.072167 Min: 0.670 Max: 0.682
> 23:04:11.071224 Min: 0.669 Max: 0.680
> 23:04:12.071281 Min: 0.670 Max: 0.682
> 23:04:13.071338 Min: 0.669 Max: 0.681
> 23:04:14.071395 Min: 0.670 Max: 0.681
> 23:04:15.071453 Min: 0.670 Max: 0.682
> 23:04:16.072510 Min: 0.670 Max: 0.683
> 23:04:17.071567 Min: 0.668 Max: 0.682
> 23:04:18.071624 Min: 0.669 Max: 0.682
> 23:04:19.071681 Min: 0.670 Max: 0.681
> 23:04:20.071739 Min: 0.669 Max: 0.683
> 23:04:21.071796 Min: 0.669 Max: 0.681
> 23:04:22.071853 Min: 0.671 Max: 0.681
> 23:04:23.071910 Min: 0.670 Max: 0.683
> 23:04:24.071967 Min: 0.669 Max: 0.681
> 23:04:25.072025 Min: 0.671 Max: 0.681

bemasher, 2016-11-29:
Nothing in it, try a higher gain and remeasure your squelch. Remember the squelch is meant to cut out the noise floor. Your squelch in that one looks like it should have been something like 0.7 since it only captured the noise floor.

cniesen, 2016-11-29:
I'm having a hard time getting anything.  Any chance that I'm on the wrong frequency?  With rtlamr I'm on 912Mhz.  BTW, that program creates a raw sample dump as well.  Is that of any help?

[rtlamr-samplefile.zip](https://github.com/bemasher/rtlamr/files/618222/rtlamr-samplefile.zip)

```
C:\Go\workspace\rtlamr\bin>rtlamr -msgtype=scm+ -samplefile=rtlamr-samplefile
23:53:29.878351 decode.go:82: CenterFreq: 912600155
23:53:29.880351 decode.go:83: SampleRate: 2359296
23:53:29.882351 decode.go:84: DataRate: 32768
23:53:29.883351 decode.go:85: ChipLength: 72
23:53:29.884351 decode.go:86: PreambleSymbols: 16
23:53:29.886351 decode.go:87: PreambleLength: 2304
23:53:29.887351 decode.go:88: PacketSymbols: 128
23:53:29.888351 decode.go:89: PacketLength: 18432
23:53:29.889351 decode.go:90: Preamble: 0001011010100011
23:53:29.891351 main.go:93: GainCount: 29
{Time:2016-11-28T23:53:48.666 Offset:0 Length:49152 SCM+:{ProtocolID:0x1E EndpointType:0xAB EndpointID:  70942350 Consumption:    382152 Tamper:0x4900 PacketCRC:0xDD00}}
{Time:2016-11-28T23:54:08.163 Offset:49152 Length:49152 SCM+:{ProtocolID:0x1E EndpointType:0xAB EndpointID:  70942350 Consumption:    382152 Tamper:0x4900 PacketCRC:0xDD00}}
{Time:2016-11-28T23:54:18.664 Offset:98304 Length:49152 SCM+:{ProtocolID:0x1E EndpointType:0xAB EndpointID:  70942350 Consumption:    382152 Tamper:0x4900 PacketCRC:0xDD00}}
{Time:2016-11-28T23:54:28.166 Offset:147456 Length:49152 SCM+:{ProtocolID:0x1E EndpointType:0xAB EndpointID:  70942350 Consumption:    382152 Tamper:0x4900 PacketCRC:0xDD00}}
{Time:2016-11-28T23:54:29.167 Offset:196608 Length:49152 SCM+:{ProtocolID:0x1E EndpointType:0xAB EndpointID:  69427479 Consumption:    710127 Tamper:0x4900 PacketCRC:0x0F01}}
{Time:2016-11-28T23:55:08.665 Offset:245760 Length:49152 SCM+:{ProtocolID:0x1E EndpointType:0xAB EndpointID:  70942350 Consumption:    382152 Tamper:0x4900 PacketCRC:0xDD00}}
{Time:2016-11-28T23:55:14.056 Offset:294912 Length:49152 SCM+:{ProtocolID:0x1E EndpointType:0xAB EndpointID:  69427929 Consumption:    326360 Tamper:0x4900 PacketCRC:0xEFD4}}
{Time:2016-11-28T23:55:28.666 Offset:344064 Length:49152 SCM+:{ProtocolID:0x1E EndpointType:0xAB EndpointID:  70942350 Consumption:    382152 Tamper:0x4900 PacketCRC:0xDD00}}
{Time:2016-11-28T23:55:48.168 Offset:393216 Length:49152 SCM+:{ProtocolID:0x1E EndpointType:0xAB EndpointID:  70942350 Consumption:    382152 Tamper:0x4900 PacketCRC:0xDD00}}
{Time:2016-11-28T23:56:18.168 Offset:442368 Length:49152 SCM+:{ProtocolID:0x1E EndpointType:0xAB EndpointID:  70942350 Consumption:    382152 Tamper:0x4900 PacketCRC:0xDD00}}
```

cniesen, 2016-11-29:
Is that again nothing?
[sample.zip](https://github.com/bemasher/rtlamr/files/618233/sample.zip)

`rtlcap -centerfreq=912600155 -samplerate=2359296 -tunergainmode=false -gainbyindex=29 -bytes=10M -squelch=0.7 -o sample`

bemasher, 2016-11-30:
That did have some packets in it, but they appear to be OOK messages (probably ERT). That sample was also fairly short, it wasn't the full 10MB.

The raw signal that rtlamr will dump contains only signal recorded for properly decoded messages, aka what you see in the sample file is exactly what the radio saw at the moment rtlamr decoded a packet.

cniesen, 2016-12-01:
> That did have some packets in it, but they appear to be OOK messages (probably [ERT](https://en.wikipedia.org/wiki/Encoder_receiver_transmitter)). That sample was also fairly short, it wasn't the full 10MB.

I guess it's good that they have OOK messages, right?  I can run the capture for a longer time.  At this setting it took a while to get even the small amount.  Might be because there are only three meters and that makes the airwaves quite quiet.

bemasher, 2016-12-01:
NIM's are frequency-shift keyed (FSK) not on-off keyed (OOK). A longer capture will have better chance of recording a NIM.

cniesen, 2016-12-02:
A full 10MB sample:

[sample3.txt](https://github.com/bemasher/rtlamr/files/627223/sample3.txt)

(FYI, this is a binary file. For some strange reason github didn't accept the zip file)

bemasher, 2016-12-03:
Unfortunately, there are no NIM's in that sample file. Either your meter is not configured to transmit them (even though it is capable), or it is transmitting them on a different frequency other than the one you recorded that sample at.

closed 2017-06-22 by bemasher