bemasher/rtlamr #68: A couple of questions re: Tamper Codes & Meter Updates

tjmcwiz, 2016-11-29:
First I want to thank you for this excellent project. I've been trying to isolate a problem in my house lately and this is just the ticket I needed. Well done and it runs on a PI 3 very well to boot. 

Now on to my questions:
1) Is there some documentation on the tamper codes and what they mean? I noticed that most of the meters in my area (including mine) have some sort of tamper codes set.
2) I've noticed that my consumption doesn't always increment. Sometimes the time stamps in the logs are minutes apart with the same consumption number. While that may be possible for some households I have a pump for my Koi pond that runs 24x7 (about 400 watts). Any idea of why that is? Have you seen this before? Trying to determine if I may have a faulty meter or if that is some behavior that just happens.

bemasher, 2016-11-29:
The tamper flags are usually meter specific and will inform the reader of things like rotations of the meter (users trying to remove a power meter) or other things like leaks and backflow in water meters. You'll need to find documentation specific to your meter. Appendix F in the attached document describe some of the flags used by various meters, it's the best documentation I've been able to find so far.

[Endpoint-Link Pro - Field Service Representative's Guide](https://github.com/bemasher/rtlamr/files/618062/TDC-0735-013.pdf)

Transceivers don't usually update at the same rate they transmit. For example, ERT compatible electric meters transmit once every 30 seconds but might only update their value once every 5 minutes. Meters which transmit IDM's use an 8-bit field (called `ConsumptionIntervalCount` in rtlamr) to indicate which reading is currently being transmitted and increments each time the consumption value is updated.

I've got a simple setup running on an RPi2 that listens for IDM's and dumps the relevant readings into an influxdb database. It handles missed messages by keeping track of the last received `ConsumptionIntervalCount` and back-filling any values which occurred between messages.

https://gist.github.com/bemasher/7af275c3d2fd3f02933e7990e8e27d41

kazoni, 2016-11-29:
What do you do with it from the database?  I threw something together in python that throws mine into MySQL and then I use Google's chart API to chart both "instant" usage (based on how long has elapsed since the last increment of the reading) and then one showing the meter reading over time.  I don't really like how I implemented it as it just constantly calls rtlamr with -unique over and over and over, so I'm looking for something a bit more efficient.  It draws about 40-50% cpu on an old computer running linux.

bemasher, 2016-11-30:
I'm using Grafana as a front-end to influxdb. Influxdb has the benefit that it will not grow uncontrollably, everything is written into a "retention policy" that specifies how long data will be kept. Continuous queries are used to aggregate the data into useful metrics over time for longer retention periods. For example, in mine I keep 5 minute power usage for 3 days, hourly usage for 3 months and daily usage for 5 years.

[image]

tjmcwiz, 2016-11-30:
Thanks for the information. I now have some homework to do.

closed 2016-11-30 by tjmcwiz