bemasher/rtlamr #69: IDM TransitTimeOffset (Question)

tjmcwiz, 2016-11-30:
Thanks for pointing me in the direction of the IDM packet. It is a much better packet to work with since I can tell if I received all the packets or not plus the other data contained within. 

There is one field that looks interesting called TransitTimeOffset. Do you have any documentation of what this is? I thought it could be seconds since the last packet update but it doesn't look like that it is if I compare the timestamps of the IDM messsages.

Any clues?

bemasher, 2016-11-30:
Many meters that use FHSS operate on a fixed pattern and timing, to avoid collision they usually and a small random time offset to the packet to be transmitted. I believe the `TransmitTimeOffset` field is that value, my guess would be that it is measured in milliseconds, but i'm not sure how much use that would be to you.

What would be of more use is the next channel the meter will transmit on, but I don't believe that information is given in IDM's.

tjmcwiz, 2016-12-01:
Thanks for the info. That would make sense ...

closed 2016-12-01 by tjmcwiz