bemasher/rtlamr #70: Consumption of type=13 returns less digits than on the actual meter

avdabke, 2016-12-05:
I just installed this on a Windows 10 laptop. I am using the rtl-sdr dongle. It is working ok with rtl_tcp from sdr# and this rtlamr package. I have an Itron 60W water meter and I think it has been setup to transmit only SCM and no IDM. However when I look at the actual analog water meter, it shows more digits. 
For instance, the consumption in the SCM message shows 1699. The actual meter shows 169,940.73.
Is this "trimming" happening in the rtlamr code or is this how the meter is transmitting? I did not see anything in the meter documentation available online that it can trip/truncate the meter value.
Let me know if you want more information. I don't know Go but I will surely try to poke around.
As you can see, for repeated messages even when water is flowing, I can see the actual analog meter show difference in value for even small changes (e.g. even a toilet flush), but of course the transmitted code won't budge until consumption reaches that decimal place.

avdabke, 2016-12-05:
Just an update. I did find some reference about setting up truncation in the meter reader. Apparently it is possible.
https://www.itron.com/na/-/media/itron/integration/whitepaper/factorsforwaterbillingaccuracy.pdf
Here is another link from a utility which is using this meter about how the data is interpreted:
http://www.gptx.org/home/showdocument?id=11670
So it definitely is not a bug in the code. 
I will close this issue and find some other way to get more granular / actual meter readings for the water monitoring and analytics use case I have in mind.

closed 2016-12-05 by avdabke

bemasher, 2016-12-06:
It is possible that your meter is storing the remainder of the consumption value in another field like the Tamper flags. That may be worth investigating.