bemasher/rtlamr #73: Limited output from rtlamr

beschaaflh, 2017-03-26:
I'm trying to run rtlamr on a pi zero.  I get what appears to be a valid consumption reading about once a day.  Here's recent output from the rtlamr terminal window:

Time:2017-03-25T20:14:27.534 SCM:{ID:17953523 Type: 5 Tamper:{Phy:00 Enc:03} Consumption:15845892 CRC:0xAA92}}

Time:2017-03-26T06:58:29.755 SCM:{ID:  915063 Type:11 Tamper:{Phy:00 Enc:03} Consumption:6973824 CRC:0x5561}}

Meanwhile, the terminal window running rtl_tcp appears to receive data every 62 seconds.  It shows output 11+, now 1 up to 502 (it takes about 28 seconds for the count from 1 to 502).

Any idea why I'm not getting regular more frequent rtlamr output?

bemasher, 2017-03-27:
Your Pi Zero does not appear to be able to keep up with the sample rate you've chosen. Go 1.8 has some significant performance improvements over the current binary releases available here on github, you'll want to compile from source with go1.8 if you want to give it a fighting chance. Otherwise, back off on the symbol length you're using.

It's also possible that there are no ERT compatible meters in your area and those messages you're receiving are bogus. In general if you don't get a flood of ERT messages on the console, even in sparsely populated areas, you probably don't have any ERT compatible meters nearby. It is still possible however that your Zero just can't keep up and is missing the messages.

beschaaflh, 2017-03-28:
Thanks Douglas.  Think my first step will be to try to run rtlamr on a raspberry pi.  Thanks again for the suggestions.  Brian

closed 2017-03-28 by bemasher