bemasher/rtlamr #74: Landis & Gyr Focus AXR-SD Power Meter

thomastech, 2017-04-20:
Thank you for sharing your project! I'd like to integrate it with my OpenHabian system to monitor power usage. But for initial testing I installed it on a Win10 Intel desktop.  

With SDR# I can see/hear brief data chirps every 17 seconds at 914.889MHz and 915.040MHz (simultaneously).   But no data is dumped while in rtlamr. 

I've experimented with the msgtype and tried all 5 formats. I also tried centerfreq 915040. No luck so far.

My power meter is a Landis & Gyr Focus AXR-SD, FCC ID OWS-NIC507. After some searching I found the github discussion in closed ticket #55 that explained this meter is not supported if it is using an encrypted communication method.  A confirmation would require checking the FCC ID info.

I checked the FCC database and could not find any reference to the communication protocol. Do you mind looking to see if there is hope of adding it as a supported meter? Here's a direct link:
https://fccid.io/OWS-NIC507

- Thomas

bemasher, 2017-04-21:
If you'd like to investigate the possibility of adding support, I'll need some signal samples. Take a look at this tool:
https://github.com/bemasher/rtlcap

In particular this section:
https://github.com/bemasher/rtlcap#squelch

5-10 minutes worth of samples at  913.689MHz with 2.4MHz sample rate would be a good place to start. Keep in mind that the squelch level is sample rate dependent.

thomastech, 2017-04-21:
Thanks for lending a helping hand! 

I couldn't find a compiled Windows compatible version of rltcap (only linux). Do you know if it is available? If not, is there another Windows RTL or SDR# tool that will record the data you need?

In case it helps in any way, I've attached a wav file snippet from SDR# that has two sequential data burst chirps in it. It was captured at 914.8889MHz in DSB radio mode.

[SDRSharp_20170421_031254Z_914888900Hz_AF.zip](https://github.com/bemasher/rtlamr/files/945954/SDRSharp_20170421_031254Z_914888900Hz_AF.zip)

- Thomas

bemasher, 2017-04-21:
You'll need to install the go toolchain at some point to test new decoders with your meter, so you may as well install it now in order to build rtlcap from source.

thomastech, 2017-04-21:
I was afraid you were going to say that. Too bad a Windows tool is not available so I can quickly record the samples that would confirm my AXR-SD power meter is a good candidate. 

I don't want to disturb my OpenHabian system since it is actively running my home automation. I will purchase & configure another RaspberryPi to do the rtlamr investigation.  Hopefully things turn out well and this isn't an unnecessary purchase. Long story short, I'll be back after I have the new RPI to record the sample signals.

- Thomas

thomastech, 2017-06-14:
Sorry for delay! Here is the rtlcap file I captured on a new Raspberry Pi 3 Model B that I purchased specifically for this investigation:
[rtlcap.zip](https://github.com/bemasher/rtlamr/files/1073352/rtlcap.zip)

My DVB-T tuner had a dipole antenna tuned for 915MHz. The power meter was less than 15 feet away so RF signal strength from my Power meter was quite high. I set the -squelch so that neighboring meters were ignored, or so I believe.

Here are the rtlcap command settings used to capture the data:
sudo ./rtlcap -centerfreq=913.689M -tunergainmode=false -gainbyindex=10 -server 192.168.1.208:1234 -bytes 100000 -squelch=12.0 -o /var/log/rltcap.dat

- Thomas

bemasher, 2017-06-15:
That sir is FSK with arbitrary length packets. Not likely a good fit for rtlamr due to some early design decisions that are not easily changed. Support for your meter is not likely to happen unless you can come across some good documentation describing the protocol that I've not been able to find. FSK support is doable, and arbitrary packet length can probably worked around, but if the protocol is encrypted, I'm not interested in adding support for it.

thomastech, 2017-06-15:
Understood, thanks for looking at it.
- Thomas

pidge, 2018-02-14:
Is it possible this is 802.15.4g? Silver Springs has a [press release about the standard](https://www.silverspringnet.com/article/press-release/silver-spring-networks-successfully-completes-ieee-802-15-4g-interoperability-testing/). According to [this page](http://rfmw.em.keysight.com/wireless/helpfiles/n7610b/Content/Main/802.15.4g%20MR_FSK.htm), It also uses FSK, has a variable packet length, and is used in the right frequency range in the US.

thomastech, 2018-02-14:
The FCC docs say that the meter incorporates a 900 MHz frequency hopping i210 Mesh radio and a 2.4GHz 802.15.4 Zigbee Home Area Network (HAN radio). 
https://fccid.io/OWS-NIC507/Test-Report/FHSS-portion-of-test-report-1007997

arcooke, 2019-04-03:
Does @thomastech's information help at all?  These meters are quite common in Texas

thomastech, 2019-04-03:
@arcooke: This old ticket is officially closed but maybe your fresh post will change that :). I'm still interested in monitoring my AXR-SD Power Meter.  Sounds like you want to do that too.

My AXR-SD meter has the standard configuration port on it (as noted by the triangle depression in the clear cover). Unfortunately it does not have the optional ANSI C12.18 optical port for reading meter data (round magnetic interface).  It lacks a blinking LED too, so no way to monitor watt hours.

I suspect the RF method is the only option to monitor the power consumption data; Too bad this meter is not supported by rtlamr.

- Thomas

Jaharmi, 2019-04-20:
I’d also be interested in whatever can be done with the Focus AXR-SD. It’s in use by the utility in my area of Southeastern Pennsylvania.

sogorman, 2019-06-19:
I have the Focus AXRe-SD here with APS in Arizona. Looking to help read the meter, machine and SDR are just chomping at the bit. Let me know what I can do to help.

douglasg14b, 2019-07-18:
Also have this power meter, all over my part of Oregon. Would love to get realtime data that my utility doesn't provide.

What info could I provide to help?

jordanm78, 2019-08-15:
We have this meter in California too.  It seems likely that it uses 802.15.4g.  The 802.15.4g standard is established by IEEE, and they have published it here:  https://ieeexplore.ieee.org/document/6486030.  It costs $33 for non IEEE members.

DongDanBam, 2019-08-24:
bemasher - Here is capture from the Focus AXR-SD meters using rclcap

The settings for this capture were -tunergainmode=false -gainbyindex=29 -squelch=0.662

[Focus AXR-SD.txt](https://github.com/bemasher/rtlamr/files/3537263/Focus.AXR-SD.txt)

bemasher, 2019-08-24:
What center frequency was this capture made at?

DongDanBam, 2019-08-25:
My capture freq was 913.689MHZ

TedBusto, 2019-09-01:
Here is a capture from another model of Landis+Gyr this is a FOCUS AXRe-SD Model Module: G5 Integrated FOCUS AXe. FCCID: R7PEG1R1S6. This capture is taken at 921.6Mhz With the command: rtlcap -centerfreq=921.6M -tunergainmode=false -gainbyindex=29 -squelch=0.703 -o=rtlcap921600hz.txt

file:///home/michael/AMR/rtlcap921600hz.txt

bemasher, 2019-09-02:
@TedBusto, unfortunately, there is nothing of use in that sample file, there are two signal pulses that may be messages but they are too weak to do anything useful with. Your meter may be transmitting outside the frequency you chose to receive at and those are spurious receptions, you may try a different center frequency.

ComplexSysSolutions, 2019-09-09:
Thanks for developing this. It's working great for my gas and water meter, but I'm having the same issue as people above with the my Landis & Gyr Focus AXR-SD.

I'm happy to sweep the the frequencies to collect data. Couple of questions:
- At what frequency increment should I sweep at?
- Roughly how long should each sweep take? 
- Is there a tool that I can use to look at the generated txt files, to see what pulses are coming in an that frequency?

I'll restrict my self to 902 Mhz to 928 Mhz as this is the stated range of the equipment in Canada.

vincent99, 2019-09-11:
@bemasher I (and probably everyone else around us) have this same meter and just got a SDR, not sure if I did this right but the rtl cap output shows occasional large jumps in the max so I think there's a signal there?  Centered at 913.689MHz.

http://vtf.me/913_689.txt

a1j, 2019-11-05:
The power company which just installed FOCUS AXRe-SD at my location claims (in a brochure) that 
1. Meter protocol is bi-directional, however they claim that "no personal information is stored in the meter and transferred" meaning probably there is no data encryption.
2. It accumulates data and transmits once a day. Meaning it will be a challenge to record it. Need to keep SDR on for days to be able to catch anything.

a1j, 2019-11-05:
FYI https://smartenergycc.org/wp-content/uploads/2012/08/000000000001021829.pdf

douglasg14b, 2019-11-05:
[000000000001021829.pdf](https://github.com/bemasher/rtlamr/files/3810765/000000000001021829.pdf)

@a1j I'm attaching the PDF directly in case of link rot

kozoke, 2019-11-28:
Hi,
+1 for possible support of the FOCUS AXR-SD. We have them on northern California.  Really neat project.  I use home assistant and am looking for other ways to track power. 

Side note:
I do have a PG&E Rainforest Eagle (gen 1) hooked up. That works but if you push the API, it locks up.  I've found a great docker image that acts as a cloud provider (locally) so all is good. But this SDR project just looks cool. Great work!!

noneck35, 2019-12-19:
Hi All, 

How close do I need to be to get this signal?  I have the Landis+Gyr type power meter no more than 50feet from my SDR Play and RTLSDR. Neither device receives more than 1 line of data in about 8 hours. at the very least I think i should be getting many lines with as many close proximity transmitters.

Ive been reading the issues forums here and am probably missing something. Is there any pointers I can get here. 

I have the short antenna that comes with the RTLSDR. 

Thanks in advance, 
Mark

badabing2005, 2020-04-25:
Is this model workable?
Are there enough details in the FCC docs?
Another AXR-SD but FCC ID: TEB-HUNTSU864
https://fccid.io/TEB-HUNTSU864

If capture would help, I can provide some.
Thanks

bemasher, 2020-04-28:
Meters and protocols that use any modulation other than manchester-coded on-off-keying are not likely to be implemented in this project.

closed 2020-04-28 by bemasher

douglasg14b, 2020-04-29:
😢

CircuitSetup, 2020-10-21:
If anyone ever finds a solution for the AXR-SD, please let everyone here know.

bemasher, 2020-10-22:
Please do not use this issue as a meeting place to discuss meters the project does not support.

bemasher, 2021-05-27:
For the curious among you, this page has quite a lot of good information regarding Landis+Gyr meters and decoding the protocol at least some of them use.

https://wiki.recessim.com/view/Gr-smart_meters_Setup_Guide