bemasher/rtlamr #75: File input? (offline from rtl_sdr)

itdaniher, 2017-05-17:
I'm doing some offline analysis of 2MHz of bandwidth around 915MHz. I suspect I'm seeing AMR broadcasts in my data, and would like to verify this. Is there any way to pipe from a file (produced with rtl_sdr) into rtlamr?

Cheers!

bemasher, 2017-05-17:
rtlamr does not currently support signal input from a file, if you've got the go toolchain installed I can give some instruction on writing a test to feed the file to the appropriate decoder, it's maybe 15-20 lines of code to determine if you have ERT compatible messages in the file.

However, rtlamr is rather sensitive to poorly chosen sample rates, if your `sample rate / 32768` is not an integer value, you'll likely miss weak messages.

itdaniher, 2017-05-18:
My go-fu is weak, but I do indeed have the toolchain handy. No promises, but if you'd share, I'll try and make the time to put something together. Unfortunately, my input rate isn't quite right, but I can resample if needbe (don't think it'll be required).

Cheers!

bemasher, 2017-05-18:
Take a look at the following code:
https://gist.github.com/bemasher/3b80133d977d3ab7cb5ea9f2df714788

You'll need to set SampleRate and SampleFilename to match your sample data. This will test for messages from all of the supported message types. Place your sample file and the source in the same directory and execute with:

`go test -v .`

closed 2017-06-22 by bemasher