bemasher/rtlamr #77: Is the scm consumption value supposed to be static?

parkerlreed, 2017-09-27:
First off, thank you a lot for this project. My metetr LCD has been dead for a little while and this is letting me  get a value from it.

When I check my meter (and consequently the others around me) I am getting solely scm and a static consumption value (including neighbors)

I turned on my stove to add a heavy load and it has stayed at the same value. Is consumption supposed to be real time or is it some other value?

````
{Time:2017-09-27T00:53:11.771 SCM:{ID:<ID> Type: 4 Tamper:{Phy:00 Enc:00} Consumption:   26085 CRC:0xB741}}
````

bemasher, 2017-09-27:
Between messages it is possible that you will receive the same consumption value.

Meters frequently update the consumption value at a different rate than they transmit messages. In general ERT compatible power meters transmit once every 30 seconds on a random channel in their hopping pattern, but they update the consumption value every 5 minutes.

parkerlreed, 2017-09-27:
Since I posted about 17 minutes ago it's still at 26085. Checking my meter it appears I have this one (actually found the technical PDF)

http://www.smartmetereducationnetwork.com/uploads/how-to-tell-if-I-have-a-ami-dte-smart-advanced-meter/Itron%20Centron%20Meter%20Technical%20Guide1482163-201106090057150.pdf

parkerlreed, 2017-09-27:
I see page 28 mentions IDM. I'll leave it running there for a bit to see if I get a message (nothing immediately)

parkerlreed, 2017-09-27:
I am getting IDM messages

````
{Time:2017-09-27T01:33:50.967 IDM:{Preamble:0x555516A3 PacketTypeID:0x1C PacketLength:0x5C HammingCode:0xC6 ApplicationVersion:0x04 ERTType:0x07 ERTSerialNumber:  22641338 ConsumptionIntervalCount:19 ModuleProgrammingState:0xBC TamperCounters:0208007F0800 AsynchronousCounters:0x00 PowerOutageFlags:000000000000 LastConsumptionCount:3290037 DifferentialConsumptionIntervals:[3 3 2 3 3 3 1 2 2 1 2 1 2 2 2 2 3 3 3 2 3 3 3 3 2 3 3 3 3 3 2 1 2 2 4 5 4 3 8 24 3 3 3 2 1 2 2] TransmitTimeOffset:2640 SerialNumberCRC:0x2CD7 PacketCRC:0x996F}}
````

I have gotten two and that LastConsumptionCount did increase!

The ID in my original post is 39139422. Does that IDM message come from that?

parkerlreed, 2017-09-27:
After looking at scm more, that idm consumption is not my meter. I know for a fact mine is the 391. I'll leave IDM running and see if mine shows up. Thanks for the tip.

parkerlreed, 2017-09-27:
OH mine doesn't do IDM... :( 
[image]

The scm type I'm receiving is type 4.

closed 2017-09-27 by parkerlreed

parkerlreed, 2017-09-27:
So from what I'm reading I'm stuck with scm type 4 and only getting that single consumption value 

It did finally update to 26086 so it seems to be working. In the end I'm just not using a lot at once lol. Thanks again for the help.