bemasher/rtlamr #78: Other SDRs?

mntnmn, 2017-10-02:


bemasher, 2017-10-02:
Please provide some more information about what you're looking for here.

ImDroided, 2017-10-04:
Hackrf would be awesome. I can then possibly pick up my meters in the 2.4ghz range

bemasher, 2017-10-04:
That is unlikely, I don't know of any ERT compatible meters (for any commodity) that operate outside of the 915 MHz ISM band.

HackRF support would likely take the form of a spectrum server compatible with the rtl_tcp protocol for the HackRF. I'll consider it.

mntnmn, 2017-10-04:
I have looked through the code and find it tightly bound to a lot of RTL specific calls in a lot of different places. This makes me believe that an RTL simulator is the only way to do it without a major re-write.
SDRAngel can open a TCP or UDP port feeding raw I/Q or demod at any device supported bandwidth and sample rate. This was my angle on getting a pipeline to rtlamr.

ImDroided, 2017-10-04:
I have one from these guys and I guess in my area they are on 2.4 at least I cant get anything in the 900 range.

https://www.comed.com/SiteCollectionDocuments/SmartEnergy/RadioFrequencyQA.PDF

bemasher, 2017-10-04:
An rtl_tcp flavored implementation for hackrf would be fairly trivial as I've already written a package for interfacing with a hackrf, however, it will not be easily portable between linux and windows.

bluegizmo83, 2018-07-24:
Any update on this being made compatible with the HackRF on Windows?

bemasher, 2018-07-24:
Have not had time or motivation to work on a port for this given that rtl-sdr dongles are so cheap and readily available.

closed 2018-07-24 by bemasher

amm3, 2022-11-19:
> An rtl_tcp flavored implementation for hackrf would be fairly trivial as I've already written a package for interfacing with a hackrf, however, it will not be easily portable between linux and windows.

FWIW, there is hackrf implementation of rtl_tcp available in this fork of main hackrf codebase: #https://github.com/avian2/hackrf 
Unfortunately I haven't been able to get it to work with rtlamr thus far.