bemasher/rtlamr #81: Log File flag not working

vishwaslokesh, 2017-11-22:
The -logfile flag is not being accepted
Throws a error : flag provided but not defined: -logfile

Thanks

kwd42, 2017-11-22:
That option was removed  Oct'16. Is it still in the documentation somewhere?
3fef2118f30633fede3f19c621e3f95c403736c7 @bemasher bemasher committed on Oct 6, 2016

vishwaslokesh, 2017-11-22:
As the flag -format still shows up, saying the output will be written to the log file, I thought the option still exists.
Thanks for the quick response.

kwd42, 2017-11-22:
Yes, the log file in this case is where ever you happen to send Stdout.

bemasher, 2017-11-22:
The description string for `-format` is as follows since 2016-10-05, 3fef2118f30633fede3f19c621e3f95c403736c7:

`format to write log messages in: plain, csv, json, or xml`

It does not describe where messages are written, only the format they are written in. The only inaccuracy here is that they are not log messages as they are written to stdout rather than stderr.

commit 19e32367 (2017-11-23): Change format description, update help.md Fixes #81

Reword the description on -format.

Remove misleading information from help.md. Leave only information about
symbollength and it's effects on the decoder.
closed 2017-11-23 by bemasher