bemasher/rtlamr #82: Stack-fault using MSVC

gvanem, 2017-12-06:
After some troubles building with *clang-cl*, I get a stack-fault on a simple command like `rtl_433.exe -h`.
Since MSVC's *cl* is not supported it seems. But I do use MSVC' *link*. 

The culprit is the huge ` r_device devices[]` array on stack in `main()`.  
Linking with `-stack:20000000` fixes it (<a href="https://www.codecogs.com/eqnedit.php?latex=\fn_jvn&space;\tiny&space;20*10^6" target="_blank"><img src="https://latex.codecogs.com/gif.latex?\fn_jvn&space;\tiny&space;20*10^6" title="\tiny 20*10^6" /></a> bytes!!).  The default stack-size is 2MByte AFAIK.

So, can this be done differently?

bemasher, 2017-12-07:
You sir, have the wrong issue tracker. This tracker is for rtlamr, I cannot provide support for librtlsdr and friends.

closed 2017-12-07 by bemasher

gvanem, 2017-12-07:
Ops. Totally wrong place.