bemasher/rtlamr #83: Itron C2SOD

rjchu, 2017-12-17:
I’ve installed rtlamr on a Raspberry Pi3 via go but rt_tcp was NOT installed as others have indicated so I installed it manually. I’m using a RTL-SDR.com dongle and everything seems to be ok. When I run rtl_test I get the following output:

—————————
rjchu@sdr-pi:/usr/local/bin $ ./rtl_test
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
[R82XX] PLL not locked!
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
——————————

...with no further messages which from what I can see.

The problem comes when I fire up rtlamr and aside from the message banner at the beginning of rtlamr, I get nothing else - no data packets no error messages. I’ve tried specifying the message types one at a time as well as leaving this option blank and letting rtlamr try and sort things out on its own - no luck in any of these cases. I have a Iton C2SOD meeting.

Is anyone successfully decoding from this meter?

Thanks and in advance for any pointers, ideas or solutions!

-Joni-

bemasher, 2017-12-18:
Can you give an example of the message banner rtlamr displays when first run?

The Itron C2SOD is capable of transmitting both ERT and OpenWay messages, whether or not rtlamr will support this meter depends on how your utility has provisioned it. If it is not configured to transmit ERT messages, rtlamr will not receive anything from it.

If your meter is provisioned for ERT, the top of the meter's face will have a label with an endpoint type and id in the format: ## #########

rjchu, 2017-12-18:
I see “Type C2SOD” and then under that is SCE (the electric company) followed by an ID in the format ######-######. Here’s the output of rtlamr:

./rtlamr -server=127.0.0.1:1234 -samplefile=/home/rjchu/output
client accepted!
06:06:46.637649 decode.go:79: CenterFreq: 912600155
06:06:46.637916 decode.go:79: SampleRate: 2359296
06:06:46.637974 decode.go:79: DataRate: 32768
06:06:46.638025 decode.go:79: ChipLength: 72
06:06:46.638073 decode.go:79: PreambleSymbols: 21
06:06:46.638125 decode.go:79: PreambleLength: 3024
06:06:46.638168 decode.go:79: PacketSymbols: 96
06:06:46.638212 decode.go:79: PacketLength: 13824
06:06:46.638258 decode.go:79: Preamble: 111110010101001100000
06:06:46.638306 main.go:94: GainCount: 29
set freq 912600155
set sample rate 2359296
set gain mode 0

rjchu, 2017-12-18:
This isn’t my meter but while googling I found a picture someone else posted. This is identical to mine aside from the serial numbers.

http://www.createhealthyhomes.com/smart_meter_images/Trabuco_Cyn_Smart_Meter.jpg

Thanks,

-Joni-

bemasher, 2017-12-19:
Unfortunately it looks like your meter is only provisioned for OpenWay which rtlamr does not support.

closed 2017-12-19 by bemasher

bemasher, 2018-04-14:
OpenWay is used in two bands: 915 MHz and 2.4 GHz. It's ZigBee-based, so you could probably write an application to sniff the messages, I'm not willing to do so with rtlamr, the protocol is totally different from the protocols rtlamr supports. The other reason is that OpenWay in most configurations uses encryption, and would require cooperation from the utility to listen to any given meter. I'm not particularly interested in implementing protocols that use encryption because that sort of implies that the utility doesn't want unauthorized radios from listening.

It probably wouldn't be too difficult to listen to your meter using one of the various zigbee sniffers available so long as you've gotten your meter's encryption key from the utility.

TD22057, 2018-04-14:
OK - thanks.  There is no way they're going to give me a key - if they were willing to do that, they'd just let my Rainforest Eagle device connect.   I guess I'll move to plan B with something like open energy monitor.