bemasher/rtlamr #84: Unable to run on High Sierra

phubbard, 2017-12-28:
Issue: rtl_tcp not installed by 'go get'.

$GOPATH/bin contains rtlamr but rtl_tcp is not there, or anywhere else on the system. 

Details: OSX 10.13.2, Go 1.9.2 from Homebrew.

Process:
go get github.com/bemasher/rtlamr

Expected:
rtl_tcp in path

Actual: 
No errors and no sign of rtl_tcp. No luck in googling. Any help appreciated.

bemasher, 2017-12-28:
The rtlsdr tools are not installed by go get, you are responsible for providing the rtl_* tools for your particular system.

The rtltcp library provided with rtlamr only allows communication between rtlamr and the rtl_tcp spectrum server, it is not the spectrum server itself. Follow the instructions at the link below to install rtl_tcp.

```markdown
* Linux: [source and build instructions](http://sdr.osmocom.org/trac/wiki/rtl-sdr)
```

closed 2017-12-28 by bemasher

phubbard, 2017-12-30:
I'll try. Your readme says that "...the package github.com/bemasher/rtltcp, which provides a means of controlling and sampling from rtl-sdr dongles via the rtl_tcp tool. This package will be automatically downloaded and installed when getting rtlamr. The following command should be all that is required to install rtlamr."

which implies that installing rtlamr pulls rtl_tcp as a dependency, which is why I was confused.

bemasher, 2017-12-30:
The package that controls rtl_tcp is installed automatically, rtl_tcp itself is not.