bemasher/rtlamr #85: Output valid JSON

jlmcgraw, 2018-01-01:
Current JSON output isn't strictly valid, it is missing double quotes and commas

eg:

{Time:2017-12-30T18:06:13.834 SCM:{ID:28439446 Type:13 Tamper:{Phy:01 Enc:00} Consumption:   79855 CRC:0x4F4F}}

should be:

{"Time": "2017 - 12 - 30 T18: 06: 13.834", "SCM": {"ID": "28439446","Type": "13",	"Tamper": {"Phy": "01",	"Enc": "00"},"Consumption": "79855","CRC": "0x4F4F"}}

bemasher, 2018-01-01:
What value was passed to `-format` for the first output example?

jlmcgraw, 2018-01-01:
Sheesh.  Mea culpa. 

-format=json works properly  when you enable it properly  :)

closed 2018-01-01 by jlmcgraw

bemasher, 2018-01-01:
For clarification to others encountering this, default output format is go's `%+v` Printf verb, which is vaguely json-like.