bemasher/rtlamr #87: Itron AMI7 not showing up

IkeHayes, 2018-01-13:
We just built a house in a new development and I can see all of the gas meters around me (type 12) so I know the program is working but no electric meters are showing up under scm or idm msgtype. The electric meters are all Itron AMI7s which should be compatible based on the datasheet I think.

Do you have a recommendation on a good place to start troubleshooting or tweaking?

bemasher, 2018-01-13:
First step is to determine the endpoint type and id of your meter from it's labeling. The face of the meter should have a label in the format 00 000000000 where the 0's are all digits 0-9. If your label has hexadecimal digits, your meter is not likely transmitting ERT messages and is instead OpenWay.

IkeHayes, 2018-01-13:
Ah, it appears I have an OpenWay meter. That's super helpful to know that OpenWay doesn't send ERT. Is the only way an OpenWay meter communicates with ZigBee? Or is there another protocol they use for communication as well. Looks like it time to go hunting for another way to gather the data.

IkeHayes, 2018-01-14:
Just saw a similar thread in the closed issues. Thanks for the response!

closed 2018-01-14 by IkeHayes

dereitz, 2024-07-31:
Sorry to revive an old thread, but I wasn't sure where else to share this.  Has anyone tried the following repo on the AMI7 model?  From what I gather, it outputs the readings via IR which can be read and then bridged to your favorite software.  I may give this a shot, but would appreciate if others can share their experiences and/or would like to collaborate on getting it working.  Thanks!

https://github.com/joba-1/ElectricityMeter