bemasher/rtlamr #90: Slowly Increasing Memory Usage

augustrmason, 2018-02-07:
When  I have the rtltcp and rtlamr running continuously on a raspberry pi.  I am seeing a slow increase in memory usage.  After about 24 hrs I am seeing a use of around 11%.

To negate this, I saw you suggested **-symbollength=8** in another issue.  I attempted that and still saw around 8.9% usage after 24hrs.  

Ideally I'd like to run this for a long time without having to restart the rtlamr and the rtltcp.  Do you have any suggestions? Thanks for your time.

bemasher, 2018-02-08:
Can you provide version information for both rtlamr and rtl_tcp?
If you built rtlamr from source, what version of golang did you use?
Which application is slowly increasing in memory? rtlamr or rtl_tcp?

augustrmason, 2018-02-09:
Both the rtlamr and rtl_tcp were built from source on the RPi.  I am using the most recent working commit of the rtlamr from the repo, and the rtl-sdr is the most recent commit as well.  Both of them just say built from source when I check for version.  I can get commit hashes if needed.

The go version is: go1.8.4 linux/arm

The rtlamr is slowly increasing in memory over time.  I also only notice this increase when I am able to see a lot of meters at once (30+). However if I run in an area that has one meter reporting very frequently I am not seeing the issue.

bemasher, 2018-02-11:
I tried to replicate your issue on a RPi3 running go1.9.2 linux/arm and could not. The highest memory usage I saw was 1.0% and it settled there after about two hours of execution. I have another RPi3 running go1.9.2 that I use for my own monitoring that has been up and running for 42 days now and is using only 0.6% of system memory. However it is limited to a single meter, though listening to all meters it can receive shouldn't have any significant effect on memory usage.

The only place in rtlamr that has significantly sized buffers is on the input where it will store received blocks of samples but that is limited to 2MB of buffered samples. Everything else is of fixed size.

I would recommend upgrading your go installation to >=go1.9.2 and see if the behavior remains.

I have

bemasher, 2018-04-24:
Closing this issue as there have been no updates or information and I've been unable to reproduce it.

closed 2018-04-24 by bemasher