bemasher/rtlamr #91: C1SC Support

ryanhatfield, 2018-02-12:
I've been searching the internet trying to find out if my meter is supported.
The meter says it's a Schlumberger C1SC, and all the information I can find is on the C1SR.

According to the FCC reg, it operates on 917.58Mhz and has a "personality module" but doesn't explicitly say it supports ERT.

According to this [FCC document](https://fccid.io/F9CC1SA/Users-Manual/users-manual-414916.pdf), the meter transmits packets with OOK, and sounds similar in usage to the C1SR.

Any knowledge  into these meters specifically? do you have any documentation written up on how I can start sniffing out these packets? I'm not sure what tools I should use to get going.

Thanks in advance!

baileybrent, 2018-02-12:
I was just about to start working on the Itron ERT message decoding but haven't had a free minute yet.  I don't know if that protocol is the same on your Schlumberger C1SC but I suspect it is.  You can try out the rtlamr code (written in Go using rtltcp) to see if the protocol is the same.  [http://bemasher.net/rtlamr/](http://bemasher.net/rtlamr/)

Schlumberger is listed as having some support:
[https://github.com/bemasher/rtlamr/blob/master/meters.csv](https://github.com/bemasher/rtlamr/blob/master/meters.csv)

ryanhatfield, 2018-02-12:
I've been playing with rtlamr (purchased my nooelec Nesdr smart for reading my meter specifically).

I've used the following command:
`rtlcap -centerfreq=917580000 -samplerate=2359296 -tunergainmode=false -gainbyindex=0 -squelch=0.667 -o test.cap -bytes=100M`

And the resulting capture file:
[test.cap.zip](https://github.com/bemasher/rtlamr/files/1717678/test.cap.zip)

I'll help out any way I can, leme know 😄

bemasher, 2018-02-13:
Your recording had a whole lot of nothing in it. May want to try again with higher gain index.

As for C1SC support, you'll have to try the various message types rtlamr supports. First step is to look at the face of your meter to see what kind of labeling it has. If it is ERT compatible it will be obvious from the labeling.

ryanhatfield, 2018-02-13:
I realized I wasn't using rtlcap right shortly after my last post.

Here's an image of my meter
[image]

There's an FCC tag, so I know it's wireless but I don't think it's ERT compatible.

bemasher, 2018-02-14:
You're right, that does not look like an ERT label. I'd be interested in seeing any transmissions you manage to record with rtlcap.

ryanhatfield, 2018-02-14:
From the FCC sheet, I think it's using 917.58MHz. Any tips on verifying the packet I capture came from the device I want? I'll try and put the antenna right behind the meter.

After you capture transmissions with rtlcap, how do you process the signal? Do you have suggestions on how to plot it so I can look at it?

ryanhatfield, 2018-02-14:
How about this?
[test.zip](https://github.com/bemasher/rtlamr/files/1725240/test.zip)

bemasher, 2018-02-14:
The output is still awfully quiet, you may want to try again with max gain and -tunergainmode=true.

ryanhatfield, 2018-02-15:
Here's a full 25M of captures.
[longtest.zip](https://github.com/bemasher/rtlamr/files/1728157/longtest.zip)

bemasher, 2018-02-16:
Unfortunately, that file has no useful or interesting signals in it.

ryanhatfield, 2018-02-16:
Hokay,

So doing some searching around, I found the [Inspectrum](https://github.com/miek/inspectrum) utility and with that I can see there aren't any signals in the last two captures I uploaded.

Using that to change capture settings, I've finally got some signals recorded here. Most I've found look to have the same preamble, but I'm still digging around and learning more as I go.

Next I'm going to try out GNU Radio and see if I can decode some of these signals.

Anyways, here's my latest capture, including one captured signal that seems to repeat:
[longtest7.zip](https://github.com/bemasher/rtlamr/files/1732634/longtest7.zip)

jparris, 2019-03-18:
I also have a C1SC unit as suspected the transmission scheme seems to be very different. As described on page 46 of the [technical reference guide](https://www.smartmetereducationnetwork.com/uploads/how-to-tell-if-I-have-a-ami-dte-smart-advanced-meter/Itron%20Centron%20Meter%20Technical%20Guide1482163-201106090057150.pdf). 

# Transmission Scheme
The RF transmission is spread spectrum using Binary Phase Shift Keyed (BPSK) modulation. A chipping signal is used to spread the carrier and actual data is on-off-keyed (OOK). The modulator output drives the amplifier chain that provides a signal of approximately 100mW. The antenna is tuned for maximum efficiency at the carrier frequency.

The figure below shows how messages are typically transmitted. In the example, an interval packet made up of 18 intervals is transmitted every five minutes. Each interval contains a pulse count equivalent to the energy consumption for a 2.5 minute interval. The interval packet transmission represents a sliding 45 minute window of data. 

Also shown in figure below, Interval packet 0 shows that the 2.5 minute interval is transmitted in one of the 18 packets of 2.5 minute information. Interval packet 1 is transmitted 5 minutes later and the selected 2.5 minute interval of data is now transmitted as the 16th packet (numbered 15). The following packet slides the information by two 2.5 minute intervals and continues for a total of nine transmissions. The information is then no longer in the sliding 45 minute window.

[image]

bemasher, 2019-12-02:
If your meter is indeed using cellnet, then your meter is and will remain unsupported by rtlamr for the foreseeable future. Sorry.

closed 2019-12-02 by bemasher

xags, 2022-12-18:
I know that this is old but is there any chance that cellnet will ever be supported? I have a cellnet meter and an SDR I could provide captures? I'm curious more than anything if I can capture the transmissions I'm not opposed to writing a PR but unfortunately I wouldn't know how to decode the signal itself to parse relevant data..