bemasher/rtlamr #92: Spurious "Schlumberger centron C1SR" SCM reading

jlmcgraw, 2018-02-20:
A couple times now I've noticed some spurious samples from my electric meter where the reported Consumption value drastically changes and then goes back to what it was:

```
{"Time":"2018-02-19T03:02:46.604491799-05:00","Offset":0,"Length":0,"Message":{"ID":xxxxxxxx,"Type":4,"TamperPhy":0,"TamperEnc":1,"Consumption":22996,"ChecksumVal":16444}}

{"Time":"2018-02-19T03:03:23.218888829-05:00","Offset":0,"Length":0,"Message":{"ID":xxxxxxxx,"Type":4,"TamperPhy":0,"TamperEnc":1,"Consumption":2710996,"ChecksumVal":16572}}

{"Time":"2018-02-19T03:03:24.603386665-05:00","Offset":0,"Length":0,"Message":{"ID":xxxxxxxx,"Type":4,"TamperPhy":0,"TamperEnc":1,"Consumption":22996,"ChecksumVal":16444}}

```
Presumably this is an issue with the meter itself and not the rtlamr code, since the checksum is being verified by the decoder before the sample is reported, but I'm curious if anyone else has encountered this.

bemasher, 2018-02-21:
Odd, would you be willing to run rtlamr with the same flags you used with this appended to the command and send me both `c1sc.bin` and `c1sc.net`?

```
-samplefile=c1sc.bin -duration=2h > c1sc.json
```

jlmcgraw, 2018-02-21:
Unfortunately I’ve only seen this occur twice in the last 2 months.  Would it be possible to instead add code to dump the samples when the consumption value seems invalid (eg maybe > 999,999) or is very different from the last read value?

It was interesting to me that the last 3 numbers of the valid and invalid readings were the same

bemasher, 2018-02-22:
You could probably add something like that fairly easily to your instance of rtlamr. There are two loops to deal with, one handles sample blocks, and one handles decoded messages within a sample block.

https://github.com/bemasher/rtlamr/blob/master/main.go#L147
https://github.com/bemasher/rtlamr/blob/master/main.go#L168

If you want to look at messages significantly different from previous messages, scope your variable outside of L147, and compare/update it inside L168.

closed 2018-04-02 by bemasher