bemasher/rtlamr #93: Initial testing?

CowboyWay, 2018-03-16:
I just setup everything on an RPI 3 Zero W, but not certain if it's wrong settings I'm using or an incompatible meter that explains what I see. 
I ran:
rtl_tcp&
rtlamr

I see...
pi@raspberrypi:~$ rtlamr 
client accepted!
set freq 912600155
02:49:59.655744 decode.go:43: CenterFreq: 912600155
02:49:59.663004 decode.go:44: SampleRate: 2359296
02:49:59.667793 decode.go:45: DataRate: 32768
02:49:59.673577 decode.go:46: ChipLength: 72
02:49:59.676978 decode.go:47: PreambleSymbols: 21
02:49:59.692523 decode.go:48: PreambleLength: 3024
02:49:59.695322 decode.go:49: PacketSymbols: 96
02:49:59.698340 decode.go:50: PacketLength: 13824
02:49:59.700703 decode.go:51: Preamble: 111110010101001100000
02:49:59.704077 main.go:94: GainCount: 29
set sample rate 2359296
set gain mode 0

and then a while later...
...
ll+, now 289
ll+, now 290
ll+, now 291
ll+, now 292
ll+, now 293
...
Any idea what that means? It keeps spooling out with an incremented value. What do others see when everything is working and there's supported meters in the area?

For reference, this is the meter I have.
http://www.gegridsolutions.com/products/brochures/i210_family.pdf

bemasher, 2018-03-16:
Your meter may not be configured to transmit SCM (the default message type). I've not tested the RPi Zero W at full-bandwidth, so it may not be processing samples fast enough to avoid rtl_tcp buffering blocks that haven't been read yet. You may want to try the other message types and also try using a lower `-symbollength` value.

Take a look at the symbollength documentation: https://github.com/bemasher/rtlamr/blob/master/help.md

CowboyWay, 2018-03-16:
Thanks for the suggestions. I've tried all the different message types just now and they all produce similar results with lines like...  
ll+, now 64
What does that line mean?

bemasher, 2018-04-02:
Your RPi Zero W likely isn't powerful enough to run rtlamr.

closed 2018-04-02 by bemasher