bemasher/rtlamr #101: [Feature Request] Support for SoapyRemote protocol

tomswartz07, 2018-09-04:
### Problem Description
I'm hoping to get rtlamr working while using [SoapyRemote](https://github.com/pothosware/SoapyRemote) in place of rtl_tcp?

After a cursory glance at the code, the `rtl_tcp` subpackage seems fairly pluggable, so it might just be a matter of handling/switching between the two protocols. 

### Expected Result
Configuring support for SoapyRemote will allow for other SDR's like LimeSDR/Airspy/Airspy Mini/etc all working for your decoder as well.

The extended sampling rates might also help with some edge case decodes, perhaps.

### Current Result, attempt to replicate:
- Configure and start SoapyRemote, using any supported SDR device, (RTLSDR, Lime Mini, etc)
- Launch `rtlamr -server=127.0.0.1:$SOAPY_PORT` 
- RTLAMR runs, and SoapyRemote logs the that the connection was successful but no messages are decoded by rtlamr.
- Conversely, same setup with plain `rtl_tcp` works fine and messages appear regularly

bemasher, 2018-09-09:
I've begun work on a go package for interacting with a SoapySDRServer, the RPC and streaming protocol is fairly straightforward, but there will not likely be any automatic hardware selection built into rtlamr. The user will be responsible for specifying the driver they want to use if an rtlsdr is not present.

bemasher, 2019-01-27:
I've implemented the API for discovering and controlling various SDR's, but the stream protocol is fairly complex and I likely won't have time to complete it. May revisit this feature in the future.

closed 2019-01-27 by bemasher