bemasher/rtlamr #102: v0.9.0 cannot decode Neptune R900M meter

abaker, 2018-09-20:
My city recently installed a Neptune R900M at my house (FCCID: P2SR900M)

Running `rtlamr -msgtype=r900` with v0.8.0 starts decoding messages immediately, but v0.9.0 does not decode any messages

I could decode messages in d4b9855, could not decode messages in 7f30323, and `go get ...` failed on 6bfcafd so I don't know if that was the bad commit. I'm a golang noob so I may have been bisecting this incorrectly.

I uploaded the output of `rtl_sdr -f 915080000 -s 2359296 -g 0 -n 13107201 r900.bin` to [DropBox](https://www.dropbox.com/s/4g73b7pkeymt925/r900.bin.gz?dl=0)

Let me know if there is anything else I can do to help

bemasher, 2018-09-20:
That commit should fix the issue. Let me know if you have any further issues.

closed 2018-09-20 by bemasher

bemasher, 2018-09-21:
See [v0.9.1](https://github.com/bemasher/rtlamr/releases/tag/v0.9.1) for a new binary release.

commit cec57c9d (2018-09-20): Store decoder packet config in r900 parser. Fixes #102