bemasher/rtlamr #104: Neptune R900 meter read strangeness

iotarho, 2018-10-10:
Hi... I'm hitting a sticking point reading a new Neptune R900 that was recently installed (rtlamr worked flawlessly for the old one).  The meter ID is spot on, but the consumption isn't matching up to the meter read.  I've been following a few other threads trying to figure out how I might go about a fix, but thought I'd ask here before spinning wheels.  Here's an example output line for the meter:

{Time:2018-10-09T12:18:28.355 R900:{ID:1820502793 Unkn1:0xA3 NoUse: 0 BackFlow:0 Consumption: 3390349 Unkn3:0x01 Leak:14 LeakNow:2}}

Following one of the other threads, I ran the following 'troubleshooting' command:
rtl_sdr -f 915080000 -s 2359296 -g 0 -n 13107201 r900.bin

and have posted the 25MB file here:
https://drive.google.com/file/d/1qGV2UjvS6Hgvwo5XvUBBuzujz1lgxjWe/view?usp=sharing

Really appreciate any pointers on what might be going on and what I might do to investigate/potentially solve this.

bemasher, 2018-10-10:
Do you have any examples of messages logged over time?

The messages themselves are certainly valid, but the interpretation of individual fields within the message is subject to the whims of whatever engineer built the thing. There are two known formats I've seen in the wild that I've received samples for, the default in rtlamr, and a binary-coded digit flavor.

Yours may be another oddball format with the same packet structure (preamble, payload, checksum).

iotarho, 2018-10-10:
Thanks for the fast reply!  Here are messages from a few different days.  The consumption # is going the right direction--it does increment upwards from time to time--but definitely not at an even pace with the reading on the meter display:

{Time:2018-10-05T13:21:15.154 R900:{ID:1820502793 Unkn1:0xA3 NoUse: 0 BackFlow:0 Consumption: 2690139 Unkn3:0x01 Leak:14 LeakNow:2}}
{Time:2018-10-08T09:43:49.505 R900:{ID:1820502793 Unkn1:0xA3 NoUse: 0 BackFlow:0 Consumption: 3181741 Unkn3:0x01 Leak:14 LeakNow:2}}
{Time:2018-10-09T10:50:15.991 R900:{ID:1820502793 Unkn1:0xA3 NoUse: 0 BackFlow:0 Consumption: 3378023 Unkn3:0x01 Leak:14 LeakNow:2}}
{Time:2018-10-09T12:18:28.355 R900:{ID:1820502793 Unkn1:0xA3 NoUse: 0 BackFlow:0 Consumption: 3390349 Unkn3:0x01 Leak:14 LeakNow:2}}

Let me know if that's helpful/confirms what you're suspecting.. happy to do more data collection as needed!

bemasher, 2018-10-14:
Can you provide a log and sample file for more even time intervals? The following command should do the trick:

```plain
rtlamr -msgtype=r900 -filterid=<your meter's id> -samplefile=r900.bin -format=json | tee r900.log
```

A day or two worth of continuous messages and samples should be enough to figure out what's going on.

iotarho, 2019-03-07:
Okay, I'm finally having a chance to follow up on seeing why the consumption reported by rtlamr doesn't match exactly with the readout on the meter.  I'm linking to r900.bin and corresponding r900.log files.  Is there some truncation or shifting going on in reading the Consumption?  Or could this just be my mistake in understanding how this works.

Thank you for taking a look at this.

[Google drive link](https://drive.google.com/drive/folders/1kaOhP5xF3Vk_qaYNlGZqlzN1hqKBTdT1?usp=sharing)

natwilson, 2019-05-27:
I've tested with my Neptune R900 meter, and I was initially confused too but I found the consumption reported is rounded to the nearest tenth of gallon and there is no decimal point.  The screen on the physical meter is rounded to the nearest hundredth of a gallon.

bemasher, 2020-04-28:
Units and precision reported are meter-specific and is left as an exercise for the user. So long as rtlamr is reporting the data the meter transmitted accurately, I'm going to close this issue.

closed 2020-04-28 by bemasher