bemasher/rtlamr #107: Sample Receive Error Handling

bemasher, 2018-11-26:
[R]ecently rtlamr on my one machine, which connects to rtlsdr no another machine via tcp. Has been, quite randomly just start spewing [in an infinite loop]:

[08:44:12.409225 main.go:145](https://github.com/bemasher/rtlamr/blob/master/main.go#L145): Error reading samples:  EOF

I had this happen once and a while when rtlsdr had died, ie, failed to open connection to remote host. But in the last couple of cases. rtlsdr appeared ok, so I restarted rtlsdr on the host it was running on, with no success. Eventually I had to control-c out of rtlamr, and restart it.

I looked at the code but thought it odd that the for loop where this error is occurring just does a continue on an EOF. Shouldn't that cause an exit from the loop?

closed 2018-11-26 by bemasher

Legomaniac, 2018-11-26:
I have experienced this issue recently as well. Thanks for addressing it!

bemasher, 2018-11-26:
I encountered this independently and had written a partial fix for it. This commit should address all of the problems with error handling in that routine (for example: it exists now).

commit 3e1b7463 (2018-11-26): Fixes #107

Correct error handling in sample receive goroutine.
Add MsgType to json and xml formats.