bemasher/rtlamr #109: RPi zero W/RPi 1 ll+ warnings.. underpowered?

iotarho, 2019-01-24:
I've had good success using rtlamr on a Macbook and a linux-based desktop.  However when running on an RPi 1 and an RPi zero W, I'm running into the "linked list" issue, where rtl_tcp reports:
ll+, now 1
ll+, now 2
...
I still occasionally get meter reads from rtlamr, but far fewer than I do on the Macbook/desktop implementations (a few per hour, versus several per minute at my location).

It looks like at least a few people in the community are using RPis with rtlamr... is anyone successfully using the RPi 1 or RPi zero W?  Or have any of you that have encountered the ll+ streams from rtl_tcp figured out a workaround?

bemasher, 2019-01-25:
I've not done much testing on the original RPi or RPi Zero with rtlamr, but I would guess that they don't have enough cpu power to receive at rtlamr's default symbol length, I would recommend reducing symbol length. Keep in mind that using a shorter symbol length directly relates to a more narrow receive bandwidth, so you will likely receive fewer messages over time.

closed 2019-01-30 by bemasher

jpetermans, 2019-02-01:
Setting the symbollength to 48 seems to work best for on my pizero.  Changing to 56 results in a few "ll+" messages. I haven't compared captures on a more powerful cpu, but working ok for me.

val123456, 2019-12-31:
Just a late addition:  I get the same on a RPi 3 if I set -msgtype to 'all' or 'scm,r900' or 'r900'.  

If I only look for scm no errors. 

Works fine on a RPi 4.