bemasher/rtlamr #110: Please consider proper error code returns

zefie, 2019-02-25:
```08:18:27.852721 main.go:179: operr: read tcp 127.0.0.1:39610->127.0.0.1:1234: read: connection reset by peer
zefie@server:~$ echo $?
0
```
How can I distinguish this from a successful -single run?

parkerlreed, 2019-02-25:
What version are you using? Git errors out properly for me
````
[parker@stealth bin]$ ./rtlamr 
11:37:02.485655 main.go:81: Error connecting to spectrum server: dial tcp 127.0.0.1:1234: connect: connection refused
[parker@stealth bin]$ echo $?
1
````

zefie, 2019-02-25:
... Why didn't I think to test on the command line, considering how collectd was behaving with my other scripts that worked on the CLI... I have tested as you have, and it does indeed return 0 in the command environment. My apologies, I will close this.

bemasher, 2019-02-26:
Network operation errors occurring while reading from `rtl_tcp` are likely not returning appropriate exit codes because the error is logged and some tear-down occurs before rtlamr fully exits.

https://github.com/bemasher/rtlamr/blob/master/main.go#L178-L180

https://github.com/bemasher/rtlamr/blob/master/main.go#L153

https://github.com/bemasher/rtlamr/blob/master/main.go#L200-L204

https://github.com/bemasher/rtlamr/blob/master/main.go#L313

The error is not propagated up the stack to the main function, so no error is produced when rtlamr exits this way. I'll have a look at what it would take to give appropriate exit codes in this case.

commit d5f20ff8 (2019-03-19): Errors occuring in rcvr.Run() now produce appropriate exit codes. Fixes #110
closed 2019-03-19 by bemasher