bemasher/rtlamr #111: Question

gbdavidx, 2019-03-04:
Can i add a frequency that my utility bill operates on? I know the exact Mhz and its in the 400 range, but not 900

bemasher, 2019-03-04:
You sure can, though I'm not aware of any meters in the wild that use a protocol rtlamr supports that operates in the 433 MHz ISM band. The flag you're looking for is something like: `-centerfreq=400M`

```
Usage of rtlamr:
  -duration=0s: time to run for, 0 for infinite, ex. 1h5m10s
  -filterid=: display only messages matching an id in a comma-separated list of ids.
  -filtertype=: display only messages matching a type in a comma-separated list of types.
  -format=plain: decoded message output format: plain, csv, json, or xml
  -msgtype=scm: comma-separated list of message types to receive: all, scm, scm+, idm, netidm, r900 and r900bcd
  -samplefile=NUL: raw signal dump file
  -single=false: one shot execution, if used with -filterid, will wait for exactly one packet from each meter id
  -symbollength=72: symbol length in samples (8, 32, 40, 48, 56, 64, 72, 80, 88, 96)
  -unique=false: suppress duplicate messages from each meter
  -version=false: display build date and commit hash

rtltcp specific:
  -agcmode=false: enable/disable rtl agc
  -centerfreq=0: center frequency to receive on
  -directsampling=false: enable/disable direct sampling
  -freqcorrection=0: frequency correction in ppm
  -gainbyindex=0: set gain by index
  -offsettuning=false: enable/disable offset tuning
  -rtlxtalfreq=0: set rtl xtal frequency
  -samplerate=0: sample rate
  -server=127.0.0.1:1234: address or hostname of rtl_tcp instance
  -testmode=false: enable/disable test mode
  -tunergain=0: set tuner gain in dB
  -tunergainmode=false: enable/disable tuner gain
  -tunerxtalfreq=0: set tuner xtal frequency
```

closed 2019-03-04 by bemasher

gbdavidx, 2019-03-04:
it didn't seem to change...?
https://imgur.com/a/p2OvYVf (screenshot) i tried it with one hyphen too....? any ideas?

gbdavidx, 2019-03-04:
https://imgur.com/a/248LLRd is says it was set on the right but will it output once it's read? according to pge's website (my utility bill) it only outputs once a day...

gbdavidx, 2019-03-04:
Sorry, last question, will the gas number show up once it's been read? Why are there a million 11+?

I understand 11 is used for water, not gas

bemasher, 2019-03-04:
The `ll+, now #` messages mean that rtlamr is not processing as fast as rtl_tcp is producing samples. It will output messages if it detects them. I don't think your meter is supported since there aren't any ERT-compatible meters that operate in the band you're listening on.

gbdavidx, 2019-03-04:
Interesting thanks. Is this any different then using 
https://github.com/merbanan/rtl_433?

gbdavidx, 2019-03-04:
It’s odd though because my water meter and electric unit are on those lists and it wasn’t picking up anything from there either

bemasher, 2019-03-05:
Receiving from your meters is dependent both on the meters you have, and the transceivers they are paired with. Not all ERT-compatible meters actually have ERT-compatible transceivers installed. If `-msgtype=all` doesn't receive from your meters with the default `centerfreq` setting, then I would guess that your meters have transceivers for some other protocol your utilities are using.

gbdavidx, 2019-03-05:
i didn't try the msgtype=all, i plan to try that tonight,  thanks!