bemasher/rtlamr #115: Centron Itron C1SR6 Meter Question

quella, 2019-05-14:
Thank you for your easy to install and configure smart meter reading tool.  I have the product up and running on a RaspberryPi 3B+ along with a NooElec SDR dongle.  I have been running your code over night and am not able to match my smart meter to the IDs that are being captured via the tool; but I do see a few others and we are in a rather rual area.  I have tried a few different antennas and configurations, but I would think my meter's ID would be based on some string in the the meter image.  Yet, in the logs I do not get a match.  I recently enabled "all" message types instead of only "SCM" and I will look in a few more hours.  Wondering if I'm missing something here based on my meter ID.  Any help or direction.  Attached is an image and I would assume the top string or bottom would be the ID.

quella, 2019-05-14:
I believe I have my answer as I can find the smart meter when I search through the SCM+ messages captured but no SCM alone.  I will dig deeper, but all the SCM messages do not appear too map to mine unit or have some fraction of my ID, but others I can see.  Strange.  I will see if I can find it via other message types as well.  Currently filtering now on my 15507... ID and SMC+ which is working.

bemasher, 2019-05-14:
Meters don't typically transmit both SCM+ and SCM, the data you're after is likely all contained in the SCM+ messages. Is there something else you're looking for in particular that SCM+ doesn't have?

quella, 2019-05-14:
No sir, not looking for anything else than what is in the SCM+ data.  I wrongly assumed that all smart meters provided all the various message types and the SCM+ was a broadcast of additional data if the utility waited or need it.  It sounds like meters do either SCM or SCM+ message types and mine happens to be the Plus version.  I'm good with that!  Thanks for the simple to use tool and extra info!  You have educated me as well, so it is a good day.

closed 2019-05-14 by quella