bemasher/rtlamr #116: RSSI/SNR output available?

gclair, 2019-05-15:
Hi, 
 I love this utility, so thank you for it. I saw that you put in a test to calculate RSSI a few years back and I'm not seeing this or any output that looks like the RSSI value is output with the other data now. 
 
Is this still possible or was it removed?

bemasher, 2019-05-15:
It was added for a project a donor had requested. I could probably re-implement it, but it is fairly resource heavy and the results are not particularly accurate on rtl-sdr flavored radios. Note that AGC should also be disabled while using it.

Can I ask what you'd like to use it for?

gclair, 2019-05-22:
I was hoping to use it for a project to show an aggregate of neighbourhood usage and calculate the square area of the neighbourhood in relation to the radio.  If known signal X is known to be Y kilometers away,  I could roughly say (no real accuracy but ballpark), "Usage over a neighbourhood of X square kilometers in relation to the receiver ... ". As I'm new to things SDR, I was thinking that was a value I could use for that calculation..

closed 2019-06-01 by gclair